keirf/greaseweazle #96: V7 Lightning Plus misreads disks with TEC FB-502 drive

philpem, 2021-04-04:
I've encountered an odd incompatibility between the V7 Lightning Plus and the TEC FB-502 disk drive. This is a fairly early Japanese-made 5.25in single-sided drive which is sometimes seen in BBC Micro external drive boxes.

This drive works fine on a BBC Micro -- I can format the disk with Acorn DFS, and it verifies fine. However, when I hook the drive up to Greaseweazle, the resulting SCP files are unreadable. Neither HxCFE nor disk-analyse can make any sense of them.

I've attached an image of the formatted disc for reference.
[fb502_formatted_bbc.zip](https://github.com/keirf/Greaseweazle/files/6253830/fb502_formatted_bbc.zip)

keirf, 2021-04-04:
Try `gw pins 2 L` to toggle the density select pin, and then try reading.

philpem, 2021-04-04:
Sadly that didn't help, the disk image is still garbage. Although the stream does look vaguely plausible in HxC.
[image]

closed 2021-07-19 by keirf