keirf/greaseweazle #101: Bug: head cleaning does not turn the motor on

philpem, 2021-04-25:
`gw clean` does not turn the motor on when cleaning the heads.  (`gw read` works fine)

This causes issues --  some drives won't allow the head to be stepped when the motor isn't running. The more serious issue is that the head can dig into some types of cleaning disc and damage or tear them if it is "scrubbed" with the disc stationary...

keirf, 2021-04-26:
It definitely does turn the motor on, just tested it myself with a normal floppy disk. Perhaps the drive isn't recognising a disk is inserted?

philpem, 2021-05-05:
I'm suspecting this might be related to the specific drive. I've just re-tested with another drive and things worked as expected. But this full-height Shugart and a Mitsubishi are giving me grief.

They do spin fine when I use a Polar floppy drive tester/exerciser to run tests -- just that the heads are filthy and inaccessible, and so I need to use a cleaning disk.

Will look into this and update the ticket if I figure anything out. The jumper settings may be a bit weird as these drives came from a BBC Micro.

closed 2021-07-19 by keirf