keirf/greaseweazle #102: Cannot import name version

mikerm19, 2021-04-30:
I'm setting up a Greaseweazle F7 for the first time. I'm kinda new to this. I'm running Linux Mint (Ubuntu distro) and am having an issue trying to run gw.

The error I get is:
user@computer:~/Downloads/floppy$ python3 Greaseweazle/gw read test.scp
Traceback (most recent call last):
  File "Greaseweazle/gw", line 11, in <module>
    import gw
  File "/home/user/Downloads/floppy/Greaseweazle/scripts/gw.py", line 15, in <module>
    from greaseweazle import version
ImportError: cannot import name 'version' from 'greaseweazle' (/home/user/Downloads/floppy/Greaseweazle/scripts/greaseweazle/__init__.py)

I've tried it without calling python3 too.
I think I have all the prerequisites:
cc is already the newest version (4:9.3.0-1ubuntu2).
python3 is already the newest version (3.8.2-0ubuntu2).

Some output from pip3 and the setup script:
Requirement already satisfied: bitarray in /home/user/.local/lib/python3.8/site-packages (2.0.1)
Requirement already satisfied: crcmod in /home/user/.local/lib/python3.8/site-packages (1.7)
Requirement already satisfied: pyserial in /home/user/.local/lib/python3.8/site-packages (3.5)
Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (0.34.2) (tries from another thread with someone having similar issues)

I noticed that the ___init__.py that the error references is empty, is that normal?
Any suggestions on what I might be doing wrong or what I might be missing?

keirf, 2021-04-30:
Should have downloaded the release rather than cloning the repository. However it should work if you `make pysetup` in the Greaseweazle root folder.

mikerm19, 2021-04-30:
Ah yup, that was it, thank you!