keirf/greaseweazle #105: 2.88MB GUI Input limited pleae extend 4Digits

Bitcorn222, 2021-05-24:
Gui Input limited to 3 Digits999 but GW supports full 1000MB/s please extend to 4 digits.
For further easy testing Rate by moment can only be 999 at max Input. 
Also Error at rate 1000MB/s in Read by track 47 fatal error reducing rate to 750 leads to Track 67 fatal error.
Internal overflow??? Maybe also bad disk at high data rate to be checked, thanks.

ourIThome, 2021-05-24:
There are a couple of GUI wrappers, which one are you using ... You might need to speak with the GUI tool developer for this one?  Keir maintains the gw.py / gw.exe / python.

keirf, 2021-05-24:
The rate specifier is fairly misleading. All it does is specify the bit rate for creating a new HFE image (eg `gw read --rate=500 a.hfe`)

It doesn't affect the precision of flux reading across USB, so cannot affect underflow/overflow.

Also, as @ourIThome says I don't maintain the GUI, you would need to contact the author of that.

Bitcorn222, 2021-05-24:
Thanks for fast reply allright the GUI is from Don Mankin sent him a message about ive only seen Keir Frazer on top off about...
So the Rate is only the sampling rate ? I mentioned to synchronise with DataRate specified by FD Format. So i will try to reduce to less then 1000 as it would be within 2.88MB. Maybe not neccessary to use it as high as i did But truly there is a limit If I set it to 1000 can read 45 Tracks T45.1: Raw Flux (130835 flux in 689.10ms) ** FATAL ERROR: setting to 750 can read T57.1: Raw Flux (144459 flux in 759.96ms) ** FATAL ERROR: setting it to 500 or below can read full 80 Tracks. 'This is not an issue anymore maybe a nice to have a look on it, thanks

keirf, 2021-05-24:
What is the fatal error exactly, or is nothing reported?

Bitcorn222, 2021-05-24:
Console Output is only fatal error as higher the value as less tracks are readable 500 is fine more leads into fatal error without anymore description

keirf, 2021-05-24:
Try adding `--bt` to the command line. Like `gw --bt read ...`

Should give more info about the fatal error you can cut and paste here. Probably the python script has crashed.

Bitcorn222, 2021-05-24:
Command:  .\gw.exe --bt read --retries=3 --rate=1000 --rpm=300 --track=h=0-1:c=0-81 "C:\xxx\gelesene Disketten\4_mydisk.hfe"
Reading c=0-81:h=0-1 revs=3
T0.0: Raw Flux (293503 flux in 772.11ms)
...
T47.0: Raw Flux (248589 flux in 623.22ms)
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 81, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\Console.py", line 36, in run
  File "gw.py", line 87, in <module>
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\tools\read.py", line 159, in main
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\tools\util.py", line 185, in with_drive_selected
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\tools\read.py", line 107, in read_to_image
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\image\hfe.py", line 74, in emit_track
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\track.py", line 214, in __init__
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\track.py", line 257, in import_flux_data
MemoryError

Here we are

keirf, 2021-05-24:
It's an out of memory error. How much ram does the system have you're running on?

Bitcorn222, 2021-05-25:
Hi Keirf you are right i use 4GB where 2,6 are already in use. During recording it reaches only 180Mb and stops. It doesnt matter the virtual memory on SSD is 8GB but not used its content. Win10 64Bit running virtual DOS Mode.

Bitcorn222, 2021-05-25:
Allright thats the reason but command
.\gw.exe --bt read --revs=1 --retries=3 --rate=1000 --rpm=300 --track=h=0-1:c=0-81 "C:\xxx\gelesene Disketten\4_mydisk.hfe"
ends in 
T81.1: Raw Flux (107078 flux in 284.94ms)
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 81, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\Console.py", line 36, in run
  File "gw.py", line 87, in <module>
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\tools\read.py", line 159, in main
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\image\image.py", line 26, in __exit__
  File "D:\a\Greaseweazle\Greaseweazle\Greaseweazle-v0.27\scripts\greaseweazle\image\hfe.py", line 107, in get_image
struct.error: ushort format requires 0 <= number <= 0xffff

Bitcorn222, 2021-05-25:
Memory down to 640MB

keirf, 2021-05-25:
Ah.... ok.... it's not possible to represent ED data in the HFE image format as the cylinder byte length is stored as a 16-bit unsigned integer. That limits length to 65535 bytes but ED cylinder is 100000 bytes. This is not fixable. But I should give a polite error message! So I'll fix that.

keirf, 2021-05-25:
> Memory down to 640MB

Figures then that 3 revs was just on the brink of working not working, 3*640Mb ~= 2GB which is around the address space limit.

ourIThome, 2021-05-25:
Would it be a suggestion to add a request to periodically dump to file / emptying temporary storage.  With the idea of doing a resume?  Actually, maybe better to dump to .raw files, as these are dumped to files for each track and side.  And if you analyse the dump in the hxc tool, and notice you have 1 bad track, you can then re image that track again.  And then convert to hfe after?

keirf, 2021-05-25:
> Would it be a suggestion to add a request to periodically dump to file / emptying temporary storage. With the idea of doing a resume? Actually, maybe better to dump to .raw files, as these are [...]

Yes this is a good idea, using Kryoflux format and/or a new Greaseweazle flux format. It's a bit tangential to the memory usage situation.

Bitcorn222, 2021-05-25:
Uuups brought a lot of work to you thanks a lot for fast responding about possible new flux format I dont know whether it is likely to read HD MFM also would be a Idea to think about to have a base for future onto bigger amount of data processing. If this is desired. great work also Flashfloppy here in use at Amstrad CPC. Thanks again

aewatt95, 2021-07-05:
I may have found a memory leak in the optimised flux_to_bitcell function:
https://github.com/keirf/Greaseweazle/blob/8f88cd94ef56ed1bdda16397aca9d162b895ad35/scripts/c_ext/optimised.c#L87-L88

A call to PyFloat_FromDouble() generates a new reference to a Python object, which is not counted for. As such, the object is not cleaned up by any garbage collection after a return.

I tried this fix an got from around 500Mb of RAM usage down to 30Mb on ```./gw read --revs 1 --retries 0 /home/andre/buffer.img```
```
PyObject *floatClock = PyFloat_FromDouble(clock);
if (PyList_Append(time_array, floatClock) < 0){
          return NULL;
}
Py_DECREF(floatClock);
```

The same exists a few more times across the optimised C-file, but this section seems to be the main culprit for high RAM usage.

keirf, 2021-07-06:
Thanks I've probably misunderstood whether PyList_Append steals a reference. I'll confirm this and fix as necessary.

keirf, 2021-07-19:
I am now back home and will make this a priority.

keirf, 2021-07-19:
Ok I suspect that fixes this perfectly well for any modern machines with a GB or two of RAM to play with. Thanks!

closed 2021-07-19 by keirf