keirf/greaseweazle #106: ERROR: Greaseweazle is in Firmware Update Mode

rataweb, 2021-05-29:
Hi,
I'm running gw on a Windows 10 with a STM32F103C connected directly to a floppy disk (with dupont jumpers).

After programming it, all it seems that it's working good:

_D:\Apps\Greaseweazle\Greaseweazle-v0.27>gw info
_Host Tools: v0.27_
_Greaseweazle:_
_Device:   COM6_
_Model:    F1_
_Firmware: v0.27 (Update Bootloader)_
_Serial:   GW049857C00600222F140605E4_
USB Rate: Full Speed (12 Mbit/s)_

But when I run `gw read disk.scp` I get the following message:

_C:\Users\rigualva\Desktop\Greaseweazle\Greaseweazle-v0.27>gw read disk.scp_
_ERROR: Greaseweazle is in Firmware Update Mode_
_· The only available action is "gw update"_
_· For normal operation disconnect from USB and remove the Update Jumper at pins DCLK-GND_
 
I tried to upload it but the task was unsuccessfully. When I run `gw update` or `gw --bt update` it takes a lot of time and finally I need to cancel because I don't get the prompt.

¿Do you know how could I solve this issue?

Thanks and Best regards,
Shark

keirf, 2021-05-29:
Remove the jumper at dclk-gnd as directed. If it is not jumpered you might have a solder short on those pins that you need to clean up

rataweb, 2021-05-29:
Hi Keirf,
It's not jumpered. I've checked the continuity with a multimeter and it's not linked.

Thanks and best regards.
Shark

keirf, 2021-05-29:
The firmware believes the DCLK/SWCLK pin is grounded. That's unfortunate because you cannot then escape Firmware Update mode. Solution: Get another Blue Pill.

closed 2021-05-29 by keirf

rataweb, 2021-05-29:
Hi Keirf,
Do you mean that it's bricked and it's impossible to recover it?

Thanks and best regards,
Shark

keirf, 2021-05-29:
You cannot run Greaseweazle if it believes the swclk pin is pulled low. Perhaps this an MCU defect. A weird one though. Anyway perhaps you can use this board for something else or you could use Greaseweazle if you remove the swclk check in the source code and rebuild it. But apart from that: no Greaseweazle on this board.

keirf, 2021-05-29:
I assume the activity LED is flashing away when you connect the Blue Pill? That also indicates being in Update mode.

rataweb, 2021-05-29:
Hi Kierf,
Do you think if I remove R10 resistor it will work? (connected directly to a floppy disk using jumper wires)

Thanks and best regards,
Shark

keirf, 2021-05-29:
No I think there is a problem.or incompatibility with the MCU. R10 is the usb id resistor and unrelated to this problem.

rataweb, 2021-05-30:
Hi Keirf,
Not sure if it will be a Windows 10 drivers issue.
Do you know which are the correct drivers to install on Windows 10?
When I connect my Greaseweazle , Windows 10 shows the device as:

`Unknown USB Device (Device Descriptor Request Failed)`

Thanks and Best Regards,
Shark

keirf, 2021-05-30:
It must eventually identify as a COM port otherwise gw info would not work at all.