keirf/greaseweazle #107: Info not returning unless under bootload

rdpeake, 2021-05-31:
Hi, I'm a little stuck.

I've eventually managed to get firmware 0.27 installed on the bluepill via stm32 v2, however following that i'm no longer able to really do anything with the device unless i have it sitting in update firmware mode.

I can no longer connect an stm32 programmer to it, it just won't connect. `gw info` works, only when in bootloader mode, when it isn't in bootloader it just hangs. trying to run `gw update` prints that it is updating the firmware, but then seemingly does nothing else - given the update file is only kb in size, i don't expect it would take long, but it never seems to complete.

Any help on how to proceed from here?

keirf, 2021-06-01:
Is the blue pill chip CH2F103? They seem to be incompatible.

rdpeake, 2021-06-01:
Yes it is, i shall see if i can find a different blue pill, or source alternative hardware

closed 2021-07-19 by keirf