keirf/greaseweazle #109: F7 Plus USB comms errors with firmware v0.26+

keirf, 2021-06-28:
User reported USB comms issues with firmware >= v0.26. Works solidly with v0.25. All tests with tools v0.28 on Windows 10 x64.

keirf, 2021-06-28:
First theory: That this relates to Github's automated build environment. So I have made my own builds of the v0.25 and v0.26 UPD files. They are linked below. You can update to either of these within the v0.28 tools by explicitly naming the update file. For example:
```
gw update path\to\test_builds\Greaseweazle-v0.25.upd
```

[test_builds.zip](https://github.com/keirf/Greaseweazle/files/6726876/test_builds.zip)

DirkBree, 2021-06-28:
Same problem, the files in your test_builds.zip are also 1:1 identical with the files from github.

keirf, 2021-06-28:
> Same problem, the files in your test_builds.zip are also 1:1 identical with the files from github.

Well so they are! Okay, a new theory required. Possibly I will bisect between the releases so we can work out which patch is the cause.

keirf, 2021-06-28:
Here is an intermediate build just before I added support for F7 v3 boards (which was the biggest firmware change between 0.25 and 0.26). If this works, we'll try a later build; if it fails, we'll try an earlier build.

[gw-977af4.zip](https://github.com/keirf/Greaseweazle/files/6727016/gw-977af4.zip)

DirkBree, 2021-06-28:
With gw-977af4.upd my GW does nothing, it reports:
** FATAL ERROR:
Command returned garbage (01 != 00)

keirf, 2021-06-28:
Okay, very very weird. I also pulled out my F7 Plus (v2 model) and updated to gw-977af4 and tested on Win10 20H2 x64 with v0.27 tools. I cannot get it to fail at all.

keirf, 2021-06-28:
Is there any chance that power cycling your F7 Plus helps? I don't often test with the 12v power input but perhaps I should now....

keirf, 2021-06-28:
Okay here's another, earlier, intermediate version:

[gw-5279e4.zip](https://github.com/keirf/Greaseweazle/files/6727250/gw-5279e4.zip)

DirkBree, 2021-06-28:
Sorry keirf, but unfortunately the same problem, "gw info" entered here several times in a row and what comes out of it is always different.

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, v2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Not found

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Not found

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Not found

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Not found

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Not found

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, v2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.28>

keirf, 2021-06-28:
Okay an even earlier version:

[gw-e837e7.zip](https://github.com/keirf/Greaseweazle/files/6727294/gw-e837e7.zip)

DirkBree, 2021-06-28:
Looks good! 

20 times "gw info" always delivers the same and good result.
E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, v2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, v2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.28>gw info
Host Tools: v0.28
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, v2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.28>

Also 3x "gw read test.adf" work without any problem.
[image]

keirf, 2021-06-28:
Ok and in fact that version is identical to v0.25... So the very first firmware change in v0.26 series messes things up here.

keirf, 2021-06-28:
Can I check: With gw-5279e4, did you update to it (I assume using v0.25 tools, since I forgot the tools are "version locked" on update and will only update to the same reported firmware version) and then just issue gw info commands? No reads/writes/seeks or any disk type actions at all?

DirkBree, 2021-06-28:
Ok, sorry that was my mistake.
I update gw-5279e4.upd again with v0.25 tools and yes it works too, it is stable.

keirf, 2021-06-28:
Ok, are you sure you tested gw-977af4?

DirkBree, 2021-06-28:
Yes, i try'd it again and it's unstable with "gw info"!

E:\Greaseweazle\Greaseweazle-v0.25>gw update gw-977af4.upd
Updating Main Firmware to v0.25...
Done.

E:\Greaseweazle\Greaseweazle-v0.25>gw info
Host Tools: v0.25
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, version 2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.25>gw info
Host Tools: v0.25
Greaseweazle:
** FATAL ERROR:
unpack requires a buffer of 2 bytes

E:\Greaseweazle\Greaseweazle-v0.25>gw info
Host Tools: v0.25
Greaseweazle:
** FATAL ERROR:
Command returned garbage (eb != 00)

E:\Greaseweazle\Greaseweazle-v0.25>gw info
Host Tools: v0.25
Greaseweazle:
  Not found

E:\Greaseweazle\Greaseweazle-v0.25>gw info
Host Tools: v0.25
Greaseweazle:
  Device:   COM6
  Model:    F7 Plus (Ant Goffart, version 2)
  Firmware: v0.25
  Serial:   GW0036001D3538511035343631
  USB Rate: Full Speed (12 Mbit/s)

E:\Greaseweazle\Greaseweazle-v0.25>gw info
Host Tools: v0.25
Greaseweazle:
  Not found

keirf, 2021-06-28:
Okay, so, let's say with v0.25 tools you try the following (testing gw info only inbetween gw updates):
- gw update gw-977af4.upd
- gw update gw-977af4.upd
- gw update gw-5279e4.upd
- gw update gw-5279e4.upd
- gw update gw-977af4.upd

(ie. both these versions are used to update to themselves, and to each other).

Also, what is your output for `gw info --bootloader`?

DirkBree, 2021-06-28:
Let's interrupt, please, I'm no longer looking through here.
Now I have flashed the gw-5279e4.upd again and there are problems with it again.
[image]

Maybe it is because the firmware is not written correctly, see picture?
I now test the versions properly and then contact me again, if necessary I flash the GW externally first with an old firmware, then I write the new one.

keirf, 2021-06-28:
It might make sense to try a full reflash of latest GW firmware? You really need a ST-Link v2 device for this though (one of the cheap clones is okay).

DirkBree, 2021-06-28:
Now i reflash it complete with every firmware you upload before, also i wrote a short batch with txt output.
20x gw info + 5x gw read + 20 gw info with output in an txt file.

Btw, i own an st link because sometimes i build Weazles for some friends. ;)
[image]

DirkBree, 2021-06-28:
Ok, now im ready. ;)

Every firmware was testet with a batchfile with 20x "gw info" 5x "gw read test.adf" and again 20x "gw info".
TXT output from every test with every firmware is included, also my batchfile.

Your uploads and my test with 100% correct installed firmware:
Greaseweazle-v0.25.upd = ok
Greaseweazle-v0.26.upd = broken
gw-977af4.upd = broken
gw-5279e4.upd = broken
gw-e837e7.upd = ok

[GWtest.zip](https://github.com/keirf/Greaseweazle/files/6727788/GWtest.zip)

keirf, 2021-06-28:
Ok: Do other F7 devices you have work with 5279e4 on this machine? Does this F7 Plus with 5279e4 work okay for you on your laptop?

This is exceedingly weird as the 5279e4 change affects drive head seeking only. Yet simply running gw info fails, and the seeking code has not been executed at all if you have run only gw info.

DirkBree, 2021-06-28:
atm i own only one complete soldered F7, but i can complete the one on the picture above tomorrow.

For another test with the laptop i must contact my neighbour again, possible i can get the notebook also only again tomorrow.

keirf, 2021-06-28:
Okay I think we're at a stage of: Use v0.25 for now, and investigate further with more devices later.

DirkBree, 2021-06-28:
Ok, thanks for all and i complete another one tomorrow.

But I can't explain it to me even with an error on the GW. ;)

Have a nice day.

keirf, 2021-06-28:
Yes as it is I cannot explain it at all. Hence we need more data points!

DirkBree, 2021-06-28:
And thats only for my stupid Z370-HD3P. 
Btw, same problem with a fresh installed Windows 11 that didn't need the zadig driver. 

Well it's good and I'll get in touch tomorrow.

keirf, 2021-06-28:
Windows 10 doesn't need the zadig driver either. Anyway yes let's see tomorrow.

ourIThome, 2021-06-28:
I know your both doing a great job hunting... So feel free to ignore my question.  Is there a way to dump the firmware from the board to check it's matching against what is supposed to be there?

DirkBree, 2021-06-28:
I'm on the way to my bed, but I quickly put together an F1 GW with an teac floppy drive.
It's complete other hardware but it's the same problem with firmware > v0.25 like the F7 Weazle.
Before I solder another F7 together tomorrow, after a backup i try a new and clean Windows 10 installation.

G n8

keirf, 2021-06-28:
That's interesting as F1 is different chip, different USB peripheral, different firmware build. Interesting... and bizarre.

keirf, 2021-06-28:
I mean it sounds host side then, except that v0.25 -> v0.26 firmware breaks things. But there's nothing there that should break things in this way.

DirkBree, 2021-06-28:
Yes but same question for the host, what should break anything if i change the firmware from 0.25 to 0.26.
Is it a byte sequence that upsets a driver?

I try it tomorrow and now really final for today i go to bed. ;)

keirf, 2021-06-29:
> Yes but same question for the host, what should break anything if i change the firmware from 0.25 to 0.26.
> Is it a byte sequence that upsets a driver?

Indeed, there are no changes in firmware that should have this effect. I suppose try 5279e4 version on the F1 and see how that works compared with v0.25, and see if same as on F7.

DirkBree, 2021-06-29:
Hello, I dealt with the problem a little more relaxed.
There is no longer a problem with the F1, the BluePill was probably a fake which sometimes causes problems, regardless of the firmware.
With an original BluePill, the firmware v0.28 runs on my computer without any problems.

Furthermore, I finished building another F7, so there are identical problems with the firmware on my system => v0.26.

So I started a Windows PE from USB, a version with a completely different Windows 10 version, different drivers and the problem is identical (see picture).
[image]

So it remains that the Greaseweazle firmware of the F7 from firmware version 0.26 has some problem with my hardware, or my hardware with the new releases.

keirf, 2021-06-29:
Is it another F7 Plus of the same kind as the original device we were investigating?

DirkBree, 2021-06-29:
It's the same F7 Plus V2, but also i own other PCBs that i can assemble tomorrow.
Also tomorrow i can flash the firmware with my ST-Link, no problem.

Today it's impisble because it is football time. ;-)

Btw, i try'd to identify the problem with an USB analyzer.
For me as a noob it looks like there Weazle lost the complete communication.
[USB_Logfiles.zip](https://github.com/keirf/Greaseweazle/files/6734623/USB_Logfiles.zip)

I'll be in touch tomorrow

keirf, 2021-06-29:
Yes there's definitely a feeling that the F7 goes AWOL. Why only on your host I do not know. Anyway, we can have fun trying some things tomorrow (bisect the one small patch which seems to have started this behaviour; try doubling the stack; try disabling caches; etc).

Football's coming home. ;)

DirkBree, 2021-06-30:
Ok, back in business after germany lost the game, i build a F7 v1 Weazle.
[image]

Flashed with "Greaseweazle-F7-v0.26.hex".
[image]

I update it to v0.28 via "gw update", downgrade it, update it to v0.28 again and it is ok!

Any idea?
Should I get a hammer and the F7 Plus .....

keirf, 2021-06-30:
I would say so but you have two F7 Plus with same behaviour?

DirkBree, 2021-06-30:
Yes and atm i reflow one of them.
But bad soldering with a problem like that?

My next idea would be to swap the 32f730 for another one from another dealer.

But somehow I don't have the reason to believe in it because everything works fine up to firmware v0.25.

Hmmmmmm

keirf, 2021-06-30:
And yet a F7 v1 works and take away the drive connection and just think about running gw info. What you have is a F7 MCU connected direct to usb (well, connected via a ESD device on F7 Plus). But basically nothing that should make a difference between the two board types. Maybe try transplant the good F7 MCU from v1 onto a bad Plus board. Does the problem stay with the board or does it move with the MCU?

DirkBree, 2021-06-30:
Reflow changes nothing and it looks good i think.
[image]

There is no need to desolder the good working ic, i can get more 32f730 from same dealer with same date code that work in v1.
ATM there are out of stock here by me since i complete the F1 and the green F7 v2 PCB.

I buy a few stm and try it out, different pcb (green / red) I already try'd.
Even if it is illogical that some 32f730 only wants to work until a specific firmware i try it.

ourIThome, 2021-06-30:
When you get the error in command prompt, is it
1) immediate
2) and does Windows make the disconnect / reconnect in Windows when you get the problem?

Is it worth playing around with the watchdog stuff?

keirf, 2021-06-30:
It's certainly not a reflow thing, your soldering is fine. The benefit of taking a working f7 off the v1 is that you know for sure that chip works (on the v1) and of course you can do it straight away. It's just trying to bisect the problem in hardware. Similarly we can try doing it on software (by trying different firmwares).

DirkBree, 2021-07-01:
Please let us wait with other attempts to affect the firmware, I just got the same F730 that work on the F7 V1 boards.

My problem is currently the 74LS07 and some other parts are out of stock which I ordered from another dealer.
I do not want to swap these parts as long I have empty boards here.

I hope this weekend the missing components are here for some more Weazles and i can try them.

@ourIThome 
Question 1 i didn't understand.
There is an problem with one/two F7 Plus Weazles, and as described is this problem directly.

2. The F7 Plus itself never disconnect.
When a problem occurs and what I have heard (Pling Plong Sound) is, if I do not use the GW, a sound as if a USB device is removed.

Until then thanks for your patience and help and forgiveness for my bad English.
I wait for the missing components and sign up as soon as possible.

keirf, 2021-07-01:
Okay no problem. I am away for two weeks from Monday so this may wait until I'm back.

DirkBree, 2021-07-01:
That sounds good!
Great holiday and we read with further information if you're back recovered.

Also thanks for this awesome support.

keirf, 2021-07-02:
I would suggest you ship me one, but as far as we can tell this is also currently limited to your one host PC?

DirkBree, 2021-07-11:
We can try that if the two next also make problems with this pc.
I now have components from another dealer here and build two completely new on other boards. I still lack the voltare regulator parts, but that does not matter.

I'll contact you as soon as the Boards are ready.

DirkBree, 2021-07-17:
So, I finally have another F7 Plus board ready at least as far as I can use it.
I am still missing components but they are not important (12V Reg. And USB Protection), they are currently not available and as a oscillator i currently have one in THT on it.
[image]
This new one works fine even with the current firmware for hours.

I will of course still try out exactly where the problem is, but it is not a bug in the software or a general incompatibility with the new firmware and my USB controller.
I suspect something is wrong with the STM32.

In any case, thank you very much for the great support and patience, now I'll have a look at delivering the new versions to DiskChange.

Or can the old F7 modified?
For me it will be ok to use the I/O port of the activity LED for the DC signal and PC15 to GND for another identifier? ;-)

Ok, this was only a stupid idea.

Thank you again and all the best
Dirk

keirf, 2021-07-18:
That's good news for me. 😊

Regarding DC you need to bring floppy pin 34 to any F7 pin and then update src/mcu/stm32f7/board.c and floppy.c to tell the firmware about the pin assignment.

DirkBree, 2021-07-18:
Hi, 
i understand that the sources requires a modification but i didn't know how i can do it.
Is there a FAQ that i can use?
My programming knowledge is only rudimentary, I would have to read what the changes look like and and how I compile the code.
If this is in a manual I will most likely be able to implement it, as well as transfer it to newer versions.

For information about the problem G.Weazle :( because it is obviously not a bug in the firmware or incompatibility with it.
The problem with the one GW is not due to the STM32, which I replaced.
[image]

[image]

I measured all resistors and they are ok, now I will gradually replace all the other components.
It's getting strange.

Have a nice day.

DirkBree, 2021-07-18:
Some things are not exactly soldered in because I was excited, but they are technically ok!

keirf, 2021-07-18:
Could it be the USBLC6 part? Did you bypass that on your new board?

DirkBree, 2021-07-18:
Yes as described before it is bypassed on the red PCB.

But a USB protection that work with Firmware up to 0.25 and not with => 0.26?

II have no idea.
I lack the logic and I will now try everything to find the problem with one PCB.
I could equip others, but I'm kind of interested in that now. ;)

keirf, 2021-07-18:
Yes well this makes no sense overall. Perhaps track which MCUs work Vs don't work and see if the problem follows them if you swap them onto "working" board?

DirkBree, 2021-07-18:
What that struggled I miss the logic after the tests, but I will try it.
All stm32 built on all other boards work, even with others, but I will try it.

Of course, it would be simple and more meaningful that one board is easy to dispose of, but now I have to find the bug about it. ;)

BTW, do you have information about the changes in the source text information I can use? That would certainly be interesting for others.

keirf, 2021-07-19:
Here's a patch which would map floppy pin 34 to MCU pin C2 on F7 Plus boards. It should be obvious how to change the pin.

[f7_plus_rdy.zip](https://github.com/keirf/Greaseweazle/files/6840223/f7_plus_rdy.zip)

closed 2021-07-19 by keirf