keirf/greaseweazle #110: Default goal of "all" in Makefile causes build to fail violating principle of least surprise

qbarnes, 2021-07-08:
If the source is checked out, simply typing `make` in the top directory the build fails in `src/board.c` on lines 25 and 26 because the macros `MCU` and `STM32F7` are not defined.

I noticed in `.github/workflows/release.yml` that the usual build is invoked with `make dist`.  Under the `dist` target, I see that whenever `make` is invoked with goal `all`, the macro `mcu` is always set.  This in turn sets `MCU` via `FLAGS` in `Rules.mk` avoiding the `src/board.c` build problem.

A suggestion I was considering for helping someone trying to build from the source (like me) is in the definition of the `TARGETS` macro on line 5, simply move the `dist` string to the front such as:
```
TARGETS := dist all blinky clean windist mrproper f1_ocd ocd flash start serial pysetup
```
That will effectively make `dist` the default goal of the build solving the build problem.  However, I don't know Greaseweazle's build setup that well yet, so this might not be the best idea since it only solves when building on Linux.  Is this reasonable approach at all or would a more robust approach that conditionalizes the default goal based on platform (`dist` vs. `windist` be better)?  (If so, in that case better to set the `.DEFAULT_GOAL` macro instead of changing `TARGETS`.)

One question I do have about the build environment is why under the `dist` target do sequences of "make clean"s immediately followed by a "make mcu=... all blinky`?  Instead, why isn't each mcu target build split out into its own unique build subdirectory?  Was this latter approach explicitly rejected for some reason or was it just simply a side effect of the number of mcu's growing over time and no one got back to cleaning it up and doing it?

keirf, 2021-07-19:
Improving the build system is not a priority, as the whole dist can be built on my machine in a few seconds. And the official builds are all done on Github anyway.

keirf, 2021-07-19:
That said if there's some good cmake or make boilerplate to pull in that does the build-to-separate-dir stuff, I'm not averse to it.

qbarnes, 2021-07-19:
I'm well-versed in `gmake`.  (I was one of its original alpha testers starting way back 1993.)  I could make those makefile changes for building in separate directories and other cleanup for you in PRs, but I'd want to ask you a few questions to best compatible your preferred style and build environments.

I was rather surprised at the existing clean style and robustness of Greaseweazle's makefiles.  I don't typically find makefiles that polished in arbitrary github projects.  Devs just normally beat on them until they "work" then ignore them.

As far as `cmake`, I have almost no experience with it.  However, it's been on my to-do list for some time to learn, so it is something I'm interested in ramping up on.  It might be worth doing a two-phase approach where getting the changes implemented first in `gmake` then rewrite it into `cmake` (if that turns out to be a cleaner, more maintainable long-term solution).

keirf, 2021-07-19:
I built this approximate framework while developing Xen about 15-20 years ago, and I import the Rules.mk stuff into all other projects. It is "inspired" by some aspects of the Linux build system. These days I definitely simply beat on it until it "works"! It is definitely not super great but I don't want to take a week(?) out to improve it which would be better spent adding features or fixing bugs.

qbarnes, 2021-07-19:
> I built this approximate framework while developing Xen about 15-20 years ago, and I import the Rules.mk stuff into all other projects. It is "inspired" by some aspects of the Linux build system.

Ah, I see.  I thought the makefiles had some Linux flavoring in them.  (I've been a Linux kernel dev for about 15 years.)

I wouldn't want you to take time away from higher priority tasks either, but there shouldn't be much for you to do.  I'd do the changes and testing and then submit PRs.  There should be no rush, so the work could be done as a background effort on both our parts.

There's other Greaseweazle work I want to do that's higher priority (to me) is to extend `gw`'s functionality to allow it to recognize some low-level FM and MFM encoding formats to detect index and data corruption and issue track rereads in an attempt to recover the corrupted data.  I maintain and am now upstream for Tim Mann's `cw2dmk` tools (https://github.com/qbarnes/cw2dmk) that provide this functionality for Catweasel cards.  I had considered translating the necessary components of Greaseweazle's Python code to C and create a `gw2dmk` version based on `cw2dmk`, but I think that's more work and would be much more fragile and less maintainable than improving my Python coding skills and slicing open and extending `gw` to do the job.  (That's why I was attempting to build `gw` for myself and hit this build issue.)  I have my doubts that my changes could eventually be upstreamed though, so I'd have to maintain the fork unless we can figure out some hook interface for `gw` to make it extensible.

keirf, 2021-07-20:
It would be nice if GW were modular and there's some attempt to do that in the codebase, making generic codec classes, track container classes, flux classes, etc. The interfaces are unproven since they are only used by the command-line tool scripts.

This is actually my first serious Python project too. I wanted to see how it works out for larger user-space projects, where previously I would have simply used C. Its main disadvantages are obvious: very slow unless hot loops are optimised in C, and weak dynamic typing (duck typing). It has great libraries though. I'd like something similar but compiled and with strict static typing. I don't like C++. Functional languages (Haskell, OCaml) would discourage collaboration. I don't know... Google's Go?

cbrunschen, 2021-07-20:
> It would be nice if GW were modular and there's some attempt to do that in the codebase, making generic codec classes, track container classes, flux classes, etc. The interfaces are unproven since [...]

[Rust](https://www.rust-lang.org/) seems to be gaining a lot of traction.

qbarnes, 2021-07-20:
> The interfaces are unproven since they are only used by the command-line tool scripts.

As I work, I'll see if I have any refactoring suggestions.

> I wanted to see how it works out for larger user-space projects, where previously I would have simply used C. Its main disadvantages are obvious: very slow unless hot loops are optimised in C, and weak dynamic typing (duck typing).

C would have been my preferred language.  :)  Ugly, easy to blow your foot off, but gets the job done and is fast.

Yes, that's probably my single biggest complaint with Python is how incredibly slow it is compared to any compiled languages.  That's fine for things like scripting, but it really shows it's limitations when doing data processing in the language.

> I don't like C++.

I get you.  I'm "okay" with C++.  I've been writing in C++ since 1990 (Cfront 2.1 days), but still feel a long way from mastering it.  The language has come a really long way over the decades.  C++17 (and later) is not too bad as a language to write in.  As long as I'm primarily making use of the C++ standard library, I do okay, but I still struggle trying to get object design right with my own libraries written in C++.

> Functional languages (Haskell, OCaml) would discourage collaboration. I don't know... Google's Go?

I can see your point with functional languages.  I used to write in them some, but haven't used them much this century though.  They're also not just that performant for data-slinging jobs, often slower than even Python.

I'm not a fan of Go, in part because they attempted to deceive devs early on doing a hard sell job with its so-called "advantages" over other languages.  They're also trying to rewrite the world in it, and unless they've changed it in the last couple of years, it "doesn't play well with others".

Rust was mentioned.  It's a language on my "to do" list to learn.  I've played with it a little already.  One problem I have with Rust is that's it's still an rapidly evolving language.  Breaking changes are still being made.  Another is it is really hard to develop in.  You have to have the code essentially complete and fully correct to get something to even compile.  That can be a good thing when code is in maintenance mode, but frustrating challenge to experiment and prototype in.  It can be very hard to get some ideas "mostly working" just to see how they behave.  And lastly, it'll be much easier to find devs that know functional programming before Rust.

Of course, these are just my opinions.  Everyone has their own when it comes to their favorite and least favorite computer languages.  I'd prefer C or C++, mostly because I know them the best and a lot of others know them too.  As for other languages I'd have to come up to speed on, I'd pick Rust or Haskell.  But I think leaving the work in Python is good enough for now.

closed 2021-12-31 by keirf