keirf/greaseweazle #118: Reading from a 5.25" drive with --rpm 300 to an SCP file does not note rpm in header

qbarnes, 2021-08-19:
When reading from a 5.25" drive with `--rpm 300` to an SCP file, the read does not note the rpm speed in SCP header by setting bit 2 in byte 0x08.  A sample command: `$ sudo ./gw read --rpm 300 myfile.scp --drive=1`

This was tested with v0.30 while running under Fedora 34 with a Greaseweazle F7 Plus.

I'm still learning the SCP file format and I don't have a good way to test the `.scp` file under an emulator, so I'm not positive that this is a bug. However, going through the `gw` log output timing data and the raw SCP file and its flux timing data does seem to be otherwise correct for 300 RPM transfer.

keirf, 2021-08-20:
I don't set it, as I haven't found any tools that care. After all the rotation period is determined by index pulse timing information in track headers. I could try to derive the bit when writing an scp if it is in fact useful.

qbarnes, 2021-08-23:
I expect that you're reading a 300 RPM floppy as a SCP file and then write it out without ever giving the `--rpm 300` option (or write the SCP file back out using another peripheral) everything will "just work" because the two biases cancel out.

If you read a 300 RPM floppy as an SCP file by giving the `--rpm 300` option (and not having it noted in the SCP header as currently happens) and then try to write the SCP image back using a non-Greaseweazle peripheral on another platform writing to a 300 RPM drive I imagine 1 of 3 things will happen, the SCP software will:
  - Ignore the misset bit and the bit patterns will be 17% too fast in timings
  - Note the bit and scale the timings again creating 33% too fast timings
  - Check the bit and declare an error without writing since it's incompatible with its drive.

Any which way, it's not going to create the best image.

Checking the timings from a WD1791 data sheet, that controller allows up to a 25% error in timings, so 17% is enough to squeak by, but not 33%.

At least if people recorded their disks with `--rpm 300` and have issues, all they need to do to fix their image is binary edit the file and flip the bit.

If people accidentally read and then wrote their 300 RPM disks possibly at 360 RPM, they can use the very handy `scp_info.py` utility to generate a plot see if the SCP image might have timing problems.  (That's how I saw the problem for myself.)

For now I'd consider this a low priority issue since it's not an actual problem for myself.  I was just making SCP images to learn how `gw` is designed and implemented for processing floppy data.  I'm still planning to implement auto-recognition of FM/MFM/RX02 formatted diskettes (including mixed format) and then writing them out as DMK files.  I've finally taught myself enough Python now to be able to start reading the code.

closed 2022-01-10 by keirf