keirf/greaseweazle #119: Reading more than 8486 tracks (0-85) both sides, gives me a

M1kerochip, 2021-08-24:


M1kerochip, 2021-08-24:
Reading c=0-85:h=0-1 revs=5
T0.0: Raw Flux (476103 flux in 1063.80ms)
T0.1: Raw Flux (533763 flux in 1169.26ms)
....
T85.0: Raw Flux (191913 flux in 1093.07ms)
T85.1: Raw Flux (222714 flux in 1153.00ms)
** FATAL ERROR:
SCP: Too many tracks

From the output of reading a 1.44MB IBM PC disk, in a HD drive, bitcell read rate of 500, 5 revolutions, tracks 0-85, both sides.

keirf, 2021-08-24:
You can only store up to track 84.1 in an SCP image. Hence the failure is correct. Should it write out what it's got? Maybe. Perhaps it should warn instead of fatal?

M1kerochip, 2021-08-24:
Is the track limit new? I certainly used to store 86 tracks in earlier .scp images.

keirf, 2021-08-24:
Using Greaseweazle?

M1kerochip, 2021-08-24:
Yes. (I only have a GW! So, it wasn't something else!) I've several Atari ST images where there's data from tracks 0-83, so, I've read them using 84 and 85 tracks to make sure there's nothing beyond that.

M1kerochip, 2021-08-24:
The only difference was they're DS DD not DS HD disks.

keirf, 2021-08-24:
Can, for example, HxC load your old images with more tracks and you can view the disk with that many tracks?

M1kerochip, 2021-08-24:
No, seems not! I've a note with the 85/86 tracks that they're not working, and had to create them again with 84 tracks!

Well, ok then! I guess we can close this!

But, yes, I suppose, writing out the first 84 tracks would be beneficial, or like you say, warning I guess.
It's slightly annoying to get to the end, and then find you have no disk image!

keirf, 2021-08-24:
If you want more tracks properly supported, you might have better luck with Kryoflux streams. Worth a try anyway!

M1kerochip, 2021-08-24:
Saving out 0-83 tracks would probably solve my initial 'problem' 😝 

But, nah, I'm good with the SPC stream 😄

Thanks again!

closed 2021-12-31 by keirf