keirf/greaseweazle #120: Linux: Continually disconnects the greaseweazle

stuaxo, 2021-08-31:
Hi,
   @keirf I was messaging you before, but will make this a ticket so I can document the solution somewhere useful :)

Something in Linux is grabbing the USB port before greaseweazle.

At first I tried fighting modemmanager, but ultimately I've uninstalled it, not before adding a few more rules to `49-greaseweazle.rules`  [1]

lsmod shows nothing using cdc_acm
```
$ lsmod
Module                  Size  Used by
cdc_acm                36864  0
```

When I run `./gw info` it doesn't see the GW the first time, this output is in the journal:

```
$ journalctl -b -f
Aug 31 17:54:40 computer kernel: usb 1-2.1: reset full-speed USB device number 112 using xhci_hcd
Aug 31 17:54:40 computer kernel: usb 1-2.1: USB disconnect, device number 112
Aug 31 17:54:40 computer kernel: usb 1-2.1: new full-speed USB device number 113 using xhci_hcd
Aug 31 17:54:41 computer kernel: usb 1-2.1: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
Aug 31 17:54:41 computer kernel: usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 31 17:54:41 computer kernel: usb 1-2.1: Product: Greaseweazle
Aug 31 17:54:41 computer kernel: usb 1-2.1: Manufacturer: Keir Fraser
Aug 31 17:54:41 computer kernel: usb 1-2.1: SerialNumber: GW135194C04205400002101705
Aug 31 17:54:41 computer kernel: cdc_acm 1-2.1:1.0: ttyACM0: USB ACM device
Aug 31 17:54:41 computer snapd[2144]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
```

gw info, returns before everything all the journal output completes.

If I run `$ ./gw info; sleep 1; ./gw info` I get output that looks similar, but the second time it connects:

```
Aug 31 17:56:43 computer kernel: usb 1-2.1: reset full-speed USB device number 114 using xhci_hcd
Aug 31 17:56:43 computer kernel: usb 1-2.1: USB disconnect, device number 114
Aug 31 17:56:43 computer kernel: usb 1-2.1: new full-speed USB device number 115 using xhci_hcd
Aug 31 17:56:43 computer kernel: usb 1-2.1: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
Aug 31 17:56:43 computer kernel: usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 31 17:56:43 computer kernel: usb 1-2.1: Product: Greaseweazle
Aug 31 17:56:43 computer kernel: usb 1-2.1: Manufacturer: Keir Fraser
Aug 31 17:56:43 computer kernel: usb 1-2.1: SerialNumber: GW135194C04205400002101705
Aug 31 17:56:43 computer kernel: cdc_acm 1-2.1:1.0: ttyACM0: USB ACM device
Aug 31 17:56:44 computer snapd[2144]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
[ end of call to ./gw info]

Aug 31 17:57:12 computer kernel: usb 1-2.1: reset full-speed USB device number 115 using xhci_hcd
Aug 31 17:57:12 computer kernel: usb 1-2.1: USB disconnect, device number 115
Aug 31 17:57:12 computer kernel: usb 1-2.1: new full-speed USB device number 116 using xhci_hcd
Aug 31 17:57:12 computer kernel: usb 1-2.1: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
Aug 31 17:57:12 computer kernel: usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 31 17:57:12 computer kernel: usb 1-2.1: Product: Greaseweazle
Aug 31 17:57:12 computer kernel: usb 1-2.1: Manufacturer: Keir Fraser
Aug 31 17:57:12 computer kernel: usb 1-2.1: SerialNumber: GW135194C04205400002101705
Aug 31 17:57:12 computer kernel: cdc_acm 1-2.1:1.0: ttyACM0: USB ACM device
Aug 31 17:57:13 computer snapd[2144]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
```

I guess the USB device reset, and the device number increasing every time is expected ?

[70-snap.core.rules.d ](https://gist.github.com/stuaxo/b82e491a0ee0bbd813b87180f95026b5) has some stuff about ttyACM0 in it, but it's a bit hard to understand since it's all the rules together.

So, stuck for now - but will come back to this and see what else comes up.

[1]  -extra ModemManager rules, may be handy:
```
ATTRS{manufacturer}=="Keir Fraser", ATTRS{product}=="Greaseweazle", \
    ENV{ID_MM_TTY_BLACKLIST}="1"
ATTRS{manufacturer}=="Keir Fraser", ATTRS{product}=="Greaseweazle", \
    ENV{MTP_NO_PROBE}="1"
ATTRS{manufacturer}=="Keir Fraser", ATTRS{product}=="Greaseweazle", \
    ENV{ID_MM_PORT_IGNORE}="1"
ATTRS{manufacturer}=="Keir Fraser", ATTRS{product}=="Greaseweazle", \
    ENV{ID_MM_TTY_MANUAL_SCAN_ONLY}="1"
```

I also set the filtering-rules=paranoid in ModemManager, similar this:
`https://newbedev.com/udev-rules-seem-ignored-can-not-prevent-modem-manager-from-grabbing-device`

paran01d, 2021-09-20:
I seem to be having this same problem also, is it because I am trying to use via a USB hub ? I am on Ubuntu 20.04.3 LTS and 5.11.0-34-generic #36~20.04.1-Ubuntu SMP kernel if it helps. (ive only got USB-C ports on my laptop so I cant try without the hub)

stuaxo, 2021-09-20:
What CPU have you got?

I've got a Ryzen 2500u, I think it may be an AMD problem but can only find info on other Zen CPUs so far not this exact one.

I'm not going via USB C.

If you have the time it could be worth trying an entirely different distro booted via USB and seeing if that works ?

paran01d, 2021-09-20:
Nope this laptop has a core i7:

Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz

Ill have a try in a different distro as suggested and report back.

paran01d, 2021-09-20:
Ok not tried on another linux yet, but I did get my gw working fine under a windows 10 vm running under virtualbox on the same machine.

paran01d, 2021-09-20:
Also tried building from source, build was successful but the same result.

keirf, 2021-09-21:
If you can do a debug build of the firmware (or grab one from automated builds) and attach a serial USB adapter to the Greaseweazle, you may be able to see if the board itself is initiating the reset (eg. by crashing!).

paran01d, 2021-09-21:
using this method ? https://github.com/keirf/Greaseweazle/wiki/Firmware-Programming, I think i have an st-link programmer sitting around somewhere from my 3d printer.

paran01d, 2021-09-21:
I dont have usb ttl serial adapter, so ive ordered one and ill report back when it comes and I have been able to try this.

keirf, 2021-09-21:
You can actually 'gw update' to a debug firmware of the same version number if you prefer.

paran01d, 2021-09-21:
gw update currently doesnt work either with the same problem.. oh but i
never tried that under the windows vm.. ill give it a go.

keirf, 2021-09-21:
What Greaseweazle model do you have? Easiest programming method depends on that (if gw update doesn't work).

paran01d, 2021-09-21:
Doing the update under linux gives the same input/output error as any of the other commands. On my windows vm everything else seems to work fine (managed to write some working fmtowns floppies), but update fails with:

** FATAL ERROR:
Could not reopen port after mode switch

I have a v4 bought board, which according to gw info has the 0.27 firmware on it.. hold on does that mean I should be using the 0.27 software ?

paran01d, 2021-09-21:
Ok after running update a few times I can get it to update the firmware.. im now on version 0.30 built with debug=y... and im having a bit more success, it seems that every other time the commands work.. i.e. i run read and the first time i get an input/output error but the second time it works and i can read the disk.

Is there a way i can access the debug messages without a serial cable ?

keirf, 2021-09-21:
A firmware update is strongly recommended as I improved USB throughput for V4 boards. However, the software will work with existing older firmware.

If you want to re-flash a V4 board, you are recommended to use the USB-DFU method which is documented for Artery-chip-based Gotek units. You would place a jumper across 3V3-DFU and then connect to your Windows PC as usual and then run the Artery ISP software.

The debug messages in a debug build are on TXO pin at 3 mega baud. You can slow this down if need be at the top of `src/console.c`. You cannot get debug any other way, especially as in this case the main question is whether you get a crash report.

paran01d, 2021-09-21:
Ok then ill guess ill wait for my usb tty cable, thanks for the info and gw.. ill just read and write disks in my windows vm for the time being.

stuaxo, 2021-09-21:
I'll definitely watch this - my cable arrived, but free time to actually try this remains elusive.

tdaede, 2021-11-12:
So I ran into this with one computer - in my case, I found the problem was actually the power draw of the floppy spinning up browning out the USB port and causing either the microcontroller or possibly the USB host to reset (probably the former as the latter usually gives a different dmesg message). It might be worth trying powering your floppy drive from an external 5V power supply and seeing if the problem persists.

In my case, I found that some USB ports would deliver slightly more power, and it was a bit random depending on the position of the spindle motor.

stuaxo, 2021-11-14:
Thanks, I'll try a powered USB hub next time and update here, it may be a while as that stuff is in storage currently.

keirf, 2021-12-31:
Any further updates on this? Else it will get scrubbed.

stuaxo, 2022-01-05:
Cheers, no further updates as my greaseweazle is in storage so closing makes sense.

When it comes out of storage I'll give try it and update here if it still doesn't work.

stuaxo, 2025-09-16:
In the end, this turned out to be a power management issue.

Some of it, was just the laptop not being super well supported (even though HP is hardly unknown).

I noticed the issue most when I was using USB monitors, and got help from the amd devs on the Linux side (don't have details now as it was about a year ago), that made things a lot more stable (at this point, I was dealing with the same thing as the greaseweasel experienced, only more annoying as the monitor would continually turn off).

I'll give the GW another go, but reckon it's OK now.

The only time the laptop does intermittent power outs on USB connections is when the batter is very very low, and that's livable.

I've also been using `tlp` for power settings, to disable power management on the USB ports.

Everything should work the same on Linux as on Windows now.