keirf/greaseweazle #128: Dual floppy don't appear to be supported

axfelix, 2021-11-11:
I have an Epson SD-700, one of the more common and better models of dual 3.5" & 5.25" dual floppy drives, and I can't seem to get the Greaseweazel to read from the 3.5" -- even though it appears to map as Drive B, the gw binary only wants to image Drive A (the 5.25"). I already had to make my own cable to fit the molex plug off of the gw power supply so it would be nice if this could be fixed :)

tdaede, 2021-11-12:
FWIW I'm currently using Greaseweazle with two drives (separate ones, not a dual drive). Are you sure that the SD-700 is configured for IBM PC and not Shugart pinout (have you tried --drive=0,1,2)?

mistydemeo, 2021-11-17:
I've got @axfelix's drive on my desk right now so I'll answer. `--drive=1`/`B` works to access it. I also found the manual that details the jumper to switch the drive configuration: http://ftp.epson.com/desktop/SD800.TXT

I can't get anything off the 3.5" drive, though Alex says the 5.25" works, which at this point I think sounds more like a hardware problem (or a cabling problem?) than a Greaseweazle problem.

keirf, 2021-11-17:
Which Greaseweazle model is this? The F1 does not support drive bussing.

keirf, 2021-11-17:
If the Greaseweazle does support multiple drives then this is almost certainly indeed a drive hardware issue. Coincidentally I also have one of these dual drives, and the 3.5-inch drive is dead on mine, too.

mistydemeo, 2021-11-17:
It does support multiple drives, and it's happening even after I rejumpered the drive so the 3.5" drive is A, so - yeah, I think the 3.5" drive is just dead here.

keirf, 2021-11-18:
Not much more to say here then :)

closed 2021-11-18 by keirf