keirf/greaseweazle #129: AMS1117 on V4 PCB is not ceramic stable

tdaede, 2021-11-12:
There's no github for the V4 PCB so I'm filing the issue here. The regulator on the V4, the AMS1117, is attached to a ceramic bulk cap directly, but it's not rated for ceramic stability. I haven't noticed any problems yet but it could easily cause transient problems. It's fixable by either putting a resistor in series with the output cap or switching to a ceramic stable regulator.

keirf, 2021-11-12:
Good point. We borrowed this part from the Gotek schematic but I think that does use an electrolytic at the output. If I make another batch I will have to fix this. However I currently have about 900 so it won't be soon. Luckily I haven't heard anything attributable to instability. Presumably this would be triggered by load transients at the reg output? And that is fairly unlikely in the Greaseweazle design (after startup)

tdaede, 2021-11-12:
It's more likely to happen at light load. I haven't observed it personally so you're probably OK, but it can vary depending on batch and temperature so it's good to fix for the future. I just observed it when I was looking at adding trimode support.

keirf, 2021-11-12:
Interested question: What about all the smaller ceramics on the 3v3 output rail? Do they not matter so much? Due to size? Distance?

tdaede, 2021-11-12:
Distance, mostly, but they can in theory cause problems too. See section 7 of https://www.ti.com/lit/pdf/snva167

I've never seen bypass capacitors cause a problem myself, but also most designs with a high density of ceramic caps I've worked on also go with switching regulators or ceramic stable LDOs.

keirf, 2022-10-27:
Thinking about a rev of the GW V4 board. What would be the easiest fix here? A series resistor I'm guessing. Do you have a value in mind that would work well? Is this solution as good as replacing with an electrolytic? And any recommendations on exact placement schematically or in physical placement? Thanks!

tdaede, 2022-10-27:
Yeah, using a series resistor works. The exact value depends on the 1117
variant, but eg. the lm1117 wants between 0.3 and 22 ohms. A 0.47 or 1 ohm
resistor is probably good. The output cap should be placed as close to the
regulator as possible for low inductance.

Using a resistor is arguably better than an electrolytic because it won't
leak or wear out. Only other thing to watch out for is that ceramic
capacitors have lower capacitance with higher DC bias voltages. Something
to watch for if you're near the max voltage rating of your cap.

keirf, 2022-10-27:
Thanks. I assume there's no reason to prefer a particular capacitor pin for the resistor (output side vs ground side)?

tdaede, 2022-10-27:
Nope, it doesn't matter.

(off topic, did you see my email about pc88 support? if not I can make it
in the form of an issue instead)

keirf, 2022-10-27:
I don't recall your email. Always better as an issue so it can be tracked.