keirf/greaseweazle #131: Can't write scp files created by fluxengine: division by zero

tdaede, 2021-11-26:
Create a scp file in fluxengine like so:
```
./fluxengine write ibm1440 -i anything.img -d test.scp
```
Write it with greaseweazle:
```
/gw write test.scp
Writing c=0-81:h=0-1
Writing Track 0.0...** FATAL ERROR:
float division by zero
```

keirf, 2021-11-26:
For convenience can you upload your problem scp and provide a link?

keirf, 2021-11-26:
Scrub that, I reproduced it easily enough.

keirf, 2021-11-26:
Ok this is due to the unique way FluxEngine produces SCP images. Every track has five revolutions, even though at most one is populated. This confuses Greaseweazle. I will have to teach it about this layout.

tdaede, 2021-11-27:
Is that allowed by the SCP format? Another option is to change fluxengine's writer to not produce unpopulated tracks.

keirf, 2021-11-27:
I don't think it's explicitly discounted by the SCP spec. I probably shouldn't crash on it anyway.

davidgiven, 2021-12-03:
I'm fixing this at my end --- it's clearly wrong.

If I were to allow SCP files with more than five revolutions, would that cause any practical problems? It's essentially the one thing preventing me from using SCP as the primary file format...

d235j, 2021-12-03:
> If I were to allow SCP files with more than five revolutions, would that cause any practical problems? It's essentially the one thing preventing me from using SCP as the primary file format...

The SCP spec is unclear on this. It says:

> BYTE 0x05 is the number of revolutions, which is how many revolutions for each track is contained in the image.

> Starting at BYTE 0x04 are three longwords for each revolution that is stored.  Typically, a maximum of five sets of three longwords are stored using the SuperCard Pro's imaging program, but this is user defined and can vary from image to image.  Using BYTE 0x05 of the main file header, you can determine the number of sets of three longwords (one for each revolution stored).  The three longwords for each entry are described below: [...]

> ; 0005              NUMBER OF REVOLUTIONS (1-5)

That said I would recommend looking into [A2R 3.0](https://applesaucefdc.com/a2r/) as a primary file format as it is way more flexible than SCP. That said, it does not support index-to-index captures, intentionally requiring an overlap of at least a quarter of a track — this is because if the data was not written index-aligned, it is impossible to reconstruct it from an index-to-index capture without some overlap.

keirf, 2021-12-03:
I don't restrict SCP images to five revolutions maximum, and I've had no problems reported. So I would not worry about that.

commit 3583fa70 (2022-01-05): scp: More robust input parsing. Fixes handling older FluxEngine image files.
Fixes #131
closed 2022-01-05 by keirf