keirf/greaseweazle #132: Feature request: Option to set te recording RPM

merlinkv, 2021-11-28:
Hello Keirf.

It's possible to add an option to fix the recording RMP? The same option to read disks but for recording.

Thanks

keirf, 2021-11-28:
This is actually done automatically: The drive rpm is calculated and then written tracks are scaled to fit exactly into that rotation period. This is what you want, right?

merlinkv, 2021-11-28:
Hello Keirf.

What I find very interesting would be to be able to manually adjust the rotation speed in recording mode. The same thing that is done to read discs.

keirf, 2021-11-28:
What would that mean, though? Greaseweazle writes whole tracks. It sizes a track from the source image to a physical track on the destination disk. If you manually specify the rpm, what should happen?

merlinkv, 2021-11-28:
What I am looking for is a control similar to the reading process.

Let me explain: When I read a protected disk if I do not explicitly set the read speed to 300 RPM, the resulting SCP files are often invalid. However, if I set the read speed to 300 RPM then the resulting SCP files are valid for preservation.

I also have a SuperCard Pro and its disk writing process is significantly slower than Greaseweazle (about 300 RPM) resulting in more 100% functional protected disks.

I am not an expert like you, but what I comment is the result of many tests that I have done. So I think it would still be a good idea to be able to adjust the recording speed.

closed 2022-01-02 by keirf