keirf/greaseweazle #136: Lack of support for FM encoded disks

z80micro-mc, 2021-12-23:
I am trying to transfer RM 80 K single density FM and 166K mixed density disks (both 40T 48 tpi) using Greaseweazle without success.
I get similar failures with 346k mixed density disks (80T 96 tpi 5.25" or 135 tpi 3.5") again FM on track 0, mfm on other tracks.
 
If I try IBM.720 for 166k it fails to read track 0 (FM) side 0, ditto track 0 side 1 (fm).

It does seem to capture the DD MFM other tracks.

Support for RAW / IMG formats would appear to ony work with pre-defined disk types. Is there a method to add additional disk types?

How do I get it to double step 96 / 135 TPI drives to access pseudo 40T?

Thanks

Peter

z80micro-mc, 2021-12-24:
I have now attached a real 3.5" floppy drive with a 720k disk formatted in RM QD

using IBM.720   

346k 1SDD 
trk0.0 fails to read
trk0.1 fails to read

all trkn.1 fails to sind sector
     trkn.0 ok.

It looks like RM needs preferably support for img.cfg file or separate RM.SD, RM.DD, RM.QD, RM.PSD, RM.PDD
adding to the defined img types.

RM.SD    40T singlestep, 16 x FM, 128 byte sectors. headers trk0..39 on both sides and head =0, RM treats a 2S drive as 2 separate logical drives each with sector ids saying side 0.

RM.DD    40T singlestep, FM, 128 byte sectors trk0, 9 x MFM 512 byte sectors trk1--39,  headers trk0..39 on both sides and head =0, RM treats a 2S drive as 2 separate logical drives each with sector ids saying side 0.

RM.QD   80T  singlestep, FM, 128 byte sectors trk0, 9 x MFM 512 byte sectors trk1--79.  Headers trk0..79 on both sides head =0, RM treats a 2S drive as 2 separate logical drives each with sector ids saying side 0.

RM.PSD    Pseudo 40T doublestep, 16 x FM, 128 byte sectors. headers trk0..39 on both sides and head =0, RM treats a 2S drive as 2 separate logical drives each with sector ids saying side 0.

RM.PDD    Pseudo 40T doublestep, FM, 128 byte sectors trk0, 9 x MFM 512 byte sectors trk1--39.  Headers trk0..39 on both sides  head =0, RM treats a 2S drive as 2 separate logical drives each with sector ids saying side 0.

RM.NIM  80T singlestep, 9 x MFM 512 sectors.  Headers trk0..159 on both sides, head =0 - PCP/M86 

IBM-720K-FEAT seems to work but not IBM-720K-PCPM ie up and over.

Peter

keirf, 2021-12-24:
Currently only the hardcoded formats are supported, and all of these are same track format across the whole disk.

Adding new consistent-format disk formats hardcoded is quite easy. Extending this to allow mixed-format tracks will be a bit harder but probably not too bad. Allowing user configuration is future work.

qbarnes, 2021-12-24:
Earlier this year, I set out to extend `gw` functionality to support 256-byte FM and MFM primarily for TRS-80 1, 3, and 4 5.25" floppies and adding DMK file format support.  TRS-80 floppies can also be mixed FM and MFM encoded floppies.  I had made some progress on adding that support, but needed to up my Python skills and then got side-tracked on other projects.  I do intend on getting back to that work next month, but someone beats me to it I won't complain though!

I'm the current caretaker of [cw2dmk](https://github.com/qbarnes/cw2dmk).  Eventually given enough time, I'd like to translate some of its functionality to `gw`.

tdaede, 2022-10-29:
I see there is FM code in the tree. What's the status of that?

keirf, 2022-10-29:
It works. There is no way for users to specify disk/track format sof their own however, so you're limited to a few FM track formats (sector sizes, sector ids, etc) that are defined in the Python code.

keirf, 2023-01-02:
Consider this done, see #261

closed 2023-01-02 by keirf