keirf/greaseweazle #137: Tools v.034 WARNING message ?

retrobg011, 2021-12-30:
Yesterday I installed the latest firmware v1.0 with Tools v.034 windows version, on Greaseweazle v4.
The device works ok, but after each given command I get the message "WARNING: Optimized data routines not found: Run scripts setup.sh".
Previous Tools v.033 does not provide any error message.
That setup script exists in the Linux installation, I don't know why it appears here as well.
How to solve this, please ?

keirf, 2021-12-30:
Let me check it out. Maybe I broke the Windows build in this respect when I split out the firmware.

keirf, 2021-12-30:
This is broken by the latest version of setuptools I think. I'll have to untangle this and come up with a workaround.
```
2021-12-30T08:06:35.0684833Z running install
2021-12-30T08:06:35.2180788Z C:\hostedtoolcache\windows\Python\3.8.10\x86\pythonw.exe -E -c pass
2021-12-30T08:06:35.3099072Z C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
2021-12-30T08:06:35.3101010Z   warnings.warn(
2021-12-30T08:06:35.3102655Z C:\hostedtoolcache\windows\Python\3.8.10\x86\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
2021-12-30T08:06:35.3104582Z   warnings.warn(
2021-12-30T08:06:35.3105657Z TEST FAILED: ..\greaseweazle\optimised\ does NOT support .pth files
2021-12-30T08:06:35.3106742Z bad install directory or PYTHONPATH
2021-12-30T08:06:35.3107293Z 
2021-12-30T08:06:35.3107985Z You are attempting to install a package to a directory that is not
2021-12-30T08:06:35.3109297Z on PYTHONPATH and which Python does not read ".pth" files from.  The
2021-12-30T08:06:35.3110251Z installation directory you specified (via --install-dir, --prefix, or
2021-12-30T08:06:35.3111229Z the distutils default setting) was:
2021-12-30T08:06:35.3111733Z 
2021-12-30T08:06:35.3112349Z     ..\greaseweazle\optimised\
2021-12-30T08:06:35.3112773Z 
2021-12-30T08:06:35.3113540Z and your PYTHONPATH environment variable currently contains:
2021-12-30T08:06:35.3114154Z 
2021-12-30T08:06:35.3116050Z     ''
2021-12-30T08:06:35.3116532Z 
2021-12-30T08:06:35.3117322Z Here are some of your options for correcting the problem:
2021-12-30T08:06:35.3126842Z 
2021-12-30T08:06:35.3128172Z * You can choose a different installation directory, i.e., one that is
2021-12-30T08:06:35.3129275Z   on PYTHONPATH or supports .pth files
2021-12-30T08:06:35.3129843Z 
2021-12-30T08:06:35.3133481Z * You can add the installation directory to the PYTHONPATH environment
2021-12-30T08:06:35.3135807Z   variable.  (It must then also be on PYTHONPATH whenever you run
2021-12-30T08:06:35.3136963Z   Python and want to use the package(s) you are installing.)
2021-12-30T08:06:35.3137701Z 
2021-12-30T08:06:35.3138557Z * You can set up the installation directory to support ".pth" files by
2021-12-30T08:06:35.3139499Z   using one of the approaches described here:
2021-12-30T08:06:35.3140064Z 
2021-12-30T08:06:35.3141191Z   https://setuptools.pypa.io/en/latest/deprecated/easy_install.html#custom-installation-locations
2021-12-30T08:06:35.3142387Z 
2021-12-30T08:06:35.3143020Z 
2021-12-30T08:06:35.3144426Z Please make the appropriate changes for your system and try again.
```

keirf, 2021-12-30:
Okay I should be running `python -m pip install` rather than executing the `setup.py` script directly. Just going to run this through the CI workflow now.

commit 6d5965ba (2021-12-30): Windows: Fix build of Python C extension using setuptools-60.x

We need to install using pip, rather than invoking setup.py directly.

Fixes #137
closed 2021-12-30 by keirf