keirf/greaseweazle #148: Request add IBM.34 double density support

z80micro-mc, 2022-01-06:
Add IBM34 double density support as per WD279x double density system 34 track format

c=0-76:s=1-26:size=2:mfm   Standard 8" IBM system 34 MFM format

Peter

keirf, 2022-07-13:
Would/Can you test this feature?

z80micro-mc, 2022-07-13:
Kier,

Yes please, do you have a url link to the relevent version?

Peter

support (Issue #148)

keirf, 2022-07-13:
And single density ibm 3740 format would be single sided cylinders 0-76 360rpm and 26 sectors of 128 bytes per?

z80micro-mc, 2022-07-13:
> support (Issue #148)
>
> I was asking before I bother to implement it. Just to check at 360rpm
> 500k data rate and 26 sectors, how many bytes per sector I'm thinking
> 256 bytes?

Kier,

Normal IBM.34 is 256 byte sectors, but controllers support other sector 
sizes, 128, 256, 512, 1024. For RM CP/M purposes it needs 256 byte 
sectors.

It would be helpful if it could also cope with SD track 0.  Assume the 
IBM.34 table entries for various gaps, as per WD1793/2793 controller 
datasheet.

Peter

z80micro-mc, 2022-07-13:
Kier,

That is correct. Standard 8" FM disks.

RM uses 3740 format for 8" SD, and a pseudo 34 format ie normal ibm.34 
MFM except track 0 which as per RM norm is SD FM (3740)
Again gaps as per tables as per WD 1793/2793 datasheet.

Peter

support (Issue #148)

z80micro-mc, 2022-07-16:
Keir,

Correct.

Peter

keirf, 2023-01-02:
Consider this done, see #261

closed 2023-01-02 by keirf