keirf/greaseweazle #150: "gw read --track c=0-41:step=2 foo.raw" generates files with wrong numbering

markusC64, 2022-01-10:
"gw read --track c=0-41:step=2 foo.raw" generates files with wrong numbering. Compare #149  for expected numerbing.

keirf, 2022-01-10:
If you actually want to pass the tracks through with physical numbering I could add a syntax like c=0-81/2 meaning 0,2,4,...? Or perhaps c=0-81x2? Not sure on best syntax but you get the idea.

keirf, 2022-01-10:
Or you can get the same result right now by passing through gw convert afterwards with --out-tracks=step=2

markusC64, 2022-01-10:
No idea. I'm just warried to see malformed kryoflux streams spreading. Probably you'll get the same result (I cannot verify) with a 40 track floppy and kryoflux target without setting step to any value.

PS: Not for gw read, the out-tracks addition seems to work only for "gw convert"
usage: gw.exe read [options] file
gw.exe read: error: unrecognized arguments: --out-tracks=step=2

keirf, 2022-01-10:
They not thing affected here is the names of the raw files, yes. A 40 track drive would indeed name these file 0 through 39.

markusC64, 2022-01-10:
Well, a real kryoflux device does not support 40 track floppies. But 40 track disks read with a 80 track floppy is supported.

So you have to name the files the same as a 40 track floppy would do.

Well, the idea why a did that test was: People might think saving time when reading a 1541 disk where half tracks are known to be unneeded. That saving is double: While reading and while processing.

keirf, 2022-01-10:
I think I will add the new trackspec syntax I suggested. Easy to do and should solve your problem.

markusC64, 2022-01-10:
At least for 80 track drives. For 40 track drives, I don't see that it does.

keirf, 2022-01-10:
Yes then --out-tracks would be better. But I suppose noone is really using a 40t drive. They are much poorer for dumps by all accounts. So this is moot.

markusC64, 2022-01-10:
Well, I'm pretty sure about having read questions of users using a 40 track drive.

Anyway. the new trackspec syntax might do. Together with a warning about writing kryoflux filesets with largest track less than 42 - so the user knows he should check if this is on purpose or not.

It might be on purpose when dumping a single track that had errors on previous dump.

keirf, 2022-01-10:
If I allow a fractional/decimal step value then we could imagine using this for a 40t drive `tracks=c=0-81/2:step=0.5`. A bit tortuous? Means do every other track 0-81 and step half a physical track per track.

markusC64, 2022-01-10:
That would do it - but it needs a proper documentation.

Did I say I don't have a 40 track floppy to test? Well, that's true, but a flashfloppy or a hxc floppy emulator might serve as floppy for testing 40 track floppies. So I have the possibility for testing 40 track floppy via emulation.

keirf, 2022-01-11:
Okay here's a build which implements what we discussed. For example on a 40-track drive:
```--tracks=c=0-83/2:step=1/2```
Will read all cylinders and map them to 0,2,4,...

Or on an 80t drive to skip the C64 halftracks then simply:
```--tracks=c=0-83/2```

https://github.com/keirf/greaseweazle/actions/runs/1682796636

markusC64, 2022-01-11:
Thanks. I'll test that soon - probably tommorrow or the day after tommorow.

Note that I think "gw write" should be tested with all images we created in this and in the "gw convert" issue.

markusC64, 2022-01-13:
Up to now, all things that I tried worked. And no regression has been found.

closed 2022-01-29 by keirf