keirf/greaseweazle #151: gw read failes with kryoflux modified drive

markusC64, 2022-01-10:
C:\temp\gw\3>gw read --track c=0-79:h=0,1:h1.off=-8  foo.scp
Reading c=0-79:h=0-1:h1.off=-8 revs=3
T0.1: Raw Flux (136823 flux in 545.42ms)
T1.1: Raw Flux (129626 flux in 516.53ms)
T2.1: Raw Flux (128936 flux in 513.92ms)
T3.1: Raw Flux (130010 flux in 517.94ms)
T4.1: Raw Flux (129290 flux in 518.08ms)
T5.1: Raw Flux (130379 flux in 519.78ms)
T6.1: Raw Flux (130305 flux in 518.72ms)
T7.1: Raw Flux (146471 flux in 516.42ms)
** FATAL ERROR:
Track0 signal absent after seek to cylinder 0

No file is generated at all. The same command with kryoflux stream file destination also aborts with the same error, but I find the first seven tracks of the back side to be saved. Content of those files is correct.

markusC64, 2022-01-10:
Additional note: After that

gw read --track c=0-79:h=0,1  foo.scp
and
gw seek 0

fail with the same error message.

gw seek 1

repairs that so that any non negative track can be seeked to.

keirf, 2022-01-10:
This makes it appear that approaching track 0 from negative tracks, the trk0 sensor is not triggered. That seems bad to me? Like your trk0 sensor mod isn't quite right? Try this:
```
gw reset
gw seek --force -8
gw seek 0
```
compared with
```
gw reset
gw seek 8
gw seek 0
```

markusC64, 2022-01-10:
C:\temp\gw>gw reset
C:\temp\gw>gw seek --force -8
C:\temp\gw>gw seek 0
** FATAL ERROR:
Track0 signal absent after seek to cylinder 0

C:\temp\gw>gw reset
C:\temp\gw>gw seek 8
C:\temp\gw>gw seek 0

(all three with no output)

And the drive has been modified by a professional - very expensive.

The other drive mod for other drives that needs an offset of +8 for head 0 - that is reported to work. But I don't have such a drive to verify - but a trustworthy source for that information.

pin 26 shows the same result. For normal operation it's asserted at track 0 (low  active). But if coming from negative tracks, it's not asserted.

And yes, the drive's working with kryoflux without any problem at all.

keirf, 2022-01-10:
I wonder if it depends on the drive. Mine asserts track0 at track 0 and at negative tracks, whether I seek in or out. Possibly yours does not adjust the TRK0 signal when stepping inwards (since logically, how can you find track 0 when stepping inwards?!).

Out of interest what is the exact model of your flippy drive?

markusC64, 2022-01-10:
Panasonic JU-475-4 APJ

keirf, 2022-01-10:
Basically same as mine. Maybe I'll have to power mine up and have an investigation.

Anyway here is a build which may fix the issue for you. Give it a spin! Need to be logged into github to see the Artifact download link at bottom of this page:
https://github.com/keirf/greaseweazle/actions/runs/1678996851

markusC64, 2022-01-10:
Confirmed working. I have read a complete geos 2.0 disk (both sides) and written it back to the disk I cleared in the other issue. My C128 with my 1571 tells me: No error on that disk.

keirf, 2022-01-11:
This is now in master ready for v0.38 release.

closed 2022-01-11 by keirf