keirf/greaseweazle #152: To discuss: Cleared disks

markusC64, 2022-01-10:
For discussion:

So does a disk that has been cleared look in the SuperCard Pro Software (in this example Track [0):
[image]

And so does a disk look that has been cleared with the SuperCard Pro Hardware and Software:
[image]

Both with the same disk and drive taken.

PS: Kryoflux cleared disks look quite the same as greaseweazle cleared disks.

No idea why, but the SC Pro seems to clear the track different...

keirf, 2022-01-10:
SCPro uses a rapid flux pattern which reads as no fluxes at all. Greaseweazle and Kryoflux write no fluxes at all, and that results in the random flux read that you see. The latter is actually what erased media looks like (try erasing a disk with a strong magnet!).

markusC64, 2022-01-10:
I see and understand.

Anyway, I must evaulate that deeper, but there are clues that the rapid flux pattern in the half tracks of a 1541 disk do work better than no flux at all. Because from kryoflux users, I have reports that writing to 1541 disks does only work if the disk is erased with a magnet and then not cleared with kryoflux.
But with the SC Pro, I haven't observed that... Anyway, must be checked.

keirf, 2022-01-10:
Yes such an scp file could indeed be created.

keirf, 2022-01-11:
You can try this build and the new erase option `--hfreq` which should be similar to SCP's behaviour. See what you think!
(NB. Need to be logged into github to see the Artifact download link at the bottom of the page linked below):
https://github.com/keirf/greaseweazle/actions/runs/1682682090

1st-elektron, 2022-01-16:
Hi all,
i usually erase my disks with an 50Hz inductor on both sides. This method cleared even magnetic traces between the tracks.
This usually happens if your writing disk was misaligned and you try to read with an alligned disk. Even writing on this disk is a problem. 

Is it possible to erase in halftracks(160 Tracks)? Then the gaps between the tracks should be erased too.
Thank you.

markusC64, 2022-01-16:
> > Is it possible to erase in halftracks(160 Tracks)? Then the gaps between the tracks should be erased too. I would try to test this on misaligned disks. Only track 0 could be a problem if there are traces below track 0.

In general: No. But I have read detailed descriptions of a floppy modification that allows 1/4 track steps. The author has done that to allow reading disks from a 100 tpi drive in his 96 tpi drive.

closed 2022-01-29 by keirf