keirf/greaseweazle #153: Unable to read old 5.25 flopppies

raffaeler, 2022-01-25:
I bought and updated the greasewazle v4 to version 1.0
I also successfully created an img file with a 3.5" floppy.
Then I tried to read 5.25 disks: I have both DSDD(360) and DSHD(1200) flopppies that were successfully read on a very old PC last week.

Command:
```
```

keirf, 2022-01-25:
Is the 5.25 drive known good? Is it a high-density drive?

raffaeler, 2022-01-25:
Absolutely, I used it on a PC reading a lot of disks, last week.
And I tried both DSHD and DSDD floppies

raffaeler, 2022-01-25:
The drive is this one: https://www.vogonswiki.com/index.php/Teac_FD-55GFR
Is the command line correct?
Is there anything else I can do?

raffaeler, 2022-01-25:
@keirf I tried setting the pin 2 but the problem is still there.
Is the scp file useful even if I see tons of retries and missing sectors?

raffaeler, 2022-01-25:
@keirf Here is the result and log of the scp read:
[floppy.zip](https://github.com/keirf/greaseweazle/files/7934018/floppy.zip)

keirf, 2022-01-25:
Remove the `--format` option so you make a raw dump.

raffaeler, 2022-01-25:
@keirf here it is: [floppy02.zip](https://github.com/keirf/greaseweazle/files/7934162/floppy02.zip)

keirf, 2022-01-25:
Well it's ugly, the flux read is pretty messy, so no surprise the bits are unrecoverable.

Is that with pin 2 High or Low? Can you also raw-dump the opposite pin 2 setting and attach that too? As this looks a bit like a bad pin 2 setting.

raffaeler, 2022-01-25:
It was down as you asked me before.
To double check I tried to read it, but I get an error. What is the correct command to read the pin2?
```
gw pin get 2
Command Failed: GetPin: Invalid pin
```

raffaeler, 2022-01-25:
@keirf I could read a 1.2M disk by setting the pin 2 high. If I set back the pin 2 low, I only get errors.
This is a first result.

Now I still am in trouble with 360 floppies. I can't read them even turning the pin 2 low. (I used the same drive+cable+floppies that I can read from an old PC. I just re-tested the read again from the PC to double-check)
Is there any other settings to use for these ones?

keirf, 2022-01-25:
Re `gw pin get 2` you can't get the same set of pins that you can set. How weird, even I'd forgotten this! I suppose I should fix it as it's just strange otherwise.

Ok, so 1.2M now works, with pin 2 high? Great!

With 360k disks I think you need pin 2 low, and I think you need `--rpm 300` on the `gw` command line as the drive probably spins always at 360 rpm and expects the drive controller (gw in this case) to adjust for the usual 300 rpm DD speed.

raffaeler, 2022-01-25:
@keirf 

> Re `gw pin get 2` you can't get the same set of pins that you can set. How weird, even I'd forgotten this! I suppose I should fix it as it's just strange otherwise.

ok, good to know

> Ok, so 1.2M now works, with pin 2 high? Great!

Indeed!

> With 360k disks I think you need pin 2 low, and I think you need `--rpm 300` on the `gw` command line as the drive probably spins always at 360 rpm and expects the drive controller (gw in this case) to adjust for the usual 300 rpm DD speed.

It's getting better, but still I get errors and suddenly stops reading:
```
gw pin set 2 L
Pin 2 is set Low (0v)

gw.exe read --format ibm.360 --rpm=300 "C:\_floppy\mydisk.img"

Reading c=0-39:h=0-1 revs=2
Format ibm.360
T0.0: IBM MFM (9/9 sectors) from Raw Flux (82249 flux in 400.60ms)
T0.1: IBM MFM (9/9 sectors) from Raw Flux (76368 flux in 400.60ms)
T1.0: IBM MFM (0/9 sectors) from Raw Flux (69392 flux in 400.60ms)
T1.0: Retry #1
** FATAL ERROR:
Cannot append flux with different sample frequency
```

keirf, 2022-01-25:
Well that's very sad, and a bug! Apart from that this does appear to actually be working. 

Can you please run `gw.exe --bt read ...` and report the full backtrace. I may also be able to repro and will try later.

raffaeler, 2022-01-25:
@keirf  Here it is:
```
gw.exe --bt read --format ibm.360 --rpm=300 "H:\_floppy\mydisk.img"
Reading c=0-39:h=0-1 revs=2
Format ibm.360
T0.0: IBM MFM (9/9 sectors) from Raw Flux (90482 flux in 400.60ms)
T0.1: IBM MFM (9/9 sectors) from Raw Flux (86515 flux in 400.60ms)
T1.0: IBM MFM (0/9 sectors) from Raw Flux (81784 flux in 400.60ms)
T1.0: Retry #1
Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 113, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\Console.py", line 15, in run
  File "gw.py", line 98, in <module>
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.36\scripts\greaseweazle\tools\read.py", line 176, in main
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.36\scripts\greaseweazle\tools\util.py", line 212, in with_drive_selected
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.36\scripts\greaseweazle\tools\read.py", line 112, in read_to_image
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.36\scripts\greaseweazle\tools\read.py", line 61, in read_with_retry
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.36\scripts\greaseweazle\flux.py", line 38, in append
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.36\scripts\greaseweazle\error.py", line 15, in check
greaseweazle.error.Fatal: Cannot append flux with different sample frequency
```

raffaeler, 2022-01-25:
I am going to try.
I am seeing another issue. If the disk has a bad sector, the read is done until the end, but the img file is corrupt. Do you want me to open a new issue for this?

raffaeler, 2022-01-25:
Confirmed: the `--tracks=step=2` parameter is needed for the 360K (DSDD) floppy to be read correctly.
I also confirm that another disk which has a damaged sector cannot be read entirely as the bug causes gw to stop reading it.

Note. Even if the image I just read is 100% ok, the img cannot be opened with 7-zip, but I will open a different issue for this.

keirf, 2022-01-25:
Here is a build which fixes the bug when there's a damaged sector and a retry is needed. Just unzip the Windows zip file within and use these new tools. You need to be logged in to Github for the access link to work:
https://github.com/keirf/greaseweazle/suites/5058219763/artifacts/150028585

keirf, 2022-01-25:
There's also a fix (and build) in the `greaseweazle-firmware` branch which fixes the behaviour of `gw pin get`.

keirf, 2022-01-26:
@raffaeler Are you still having problems with corrupt IMG being generated?

raffaeler, 2022-01-26:
Yes, I do. I read a disk two times in a row, both times with retry set to `10`.
Firs read:
```
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ........................................
0. 2: ........................................
0. 3: ........................................
0. 4: ........................................
0. 5: ........................................
0. 6: ........................................
0. 7: ........................................
0. 8: ........................................
1. 0: X.......................................
1. 1: X.......................................
1. 2: ........................................
1. 3: ........................................
1. 4: ........................................
1. 5: .....................................X..
1. 6: ........................................
1. 7: ........................................
1. 8: ........................................
```

Second read:
```
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ........................................
0. 2: ........................................
0. 3: ........................................
0. 4: ........................................
0. 5: ........................................
0. 6: ........................................
0. 7: ........................................
0. 8: ........................................
1. 0: X.......................................
1. 1: X.......................................
1. 2: ........................................
1. 3: ........................................
1. 4: ........................................
1. 5: ........................................
1. 6: ........................................
1. 7: ........................................
1. 8: ........................................
```

Another case I suspect I depends on the seek precision is the following.
I have other cases with similar characteristics.

```
H. S: 01234567890123456789012345678901234567890123456789012345678901234567890123456789
0. 0: ................................................................................
0. 1: ................................................................................
0. 2: ................................................................................
0. 3: ................................................................................
0. 4: ................................................................................
0. 5: ................................................................................
0. 6: ................................................................................
0. 7: ................................................................................
0. 8: ................................................................................
0. 9: ................................................................................
0.10: ................................................................................
0.11: ................................................................................
0.12: ................................................................................
0.13: ................................................................................
0.14: ................................................................................
1. 0: ................................................................................
1. 1: ................................................................................
1. 2: ................................................................................
1. 3: ................................................................................
1. 4: ................................................................................
1. 5: ................................................................................
1. 6: ................................................................................
1. 7: ................................................................................
1. 8: ..............................................................................X.
1. 9: ..............................................................................X.
1.10: ....................................................................XXX.......X.
1.11: ..............................................................................X.
1.12: ..............................................................................X.
1.13: ..............................................................................XX
1.14: .........................................................................XXX..XX
```

Basically, closer the reader goes to the higher sectors, greater is the possibility of an error.

I suspect it is a seek precision problem, but I may obviously be wrong

raffaeler, 2022-01-26:
I was still with my head in the other issue :)

> @raffaeler Are you still having problems with corrupt IMG being generated?

Yes, with 7-zip I can't open all the images. I tried PeaZip and it can open all of them. 
I didn't open the new issue yet because I was trying to understand better the problem

keirf, 2022-01-26:
@raffaeler Also can you try writing a 360k disk from 360 rpm drive? I don't know how reliable that is, but I think anyway it may fail to verify because the verify is not adjusting 360rpm to 300rpm. I'm interested to know if this is the case, and if so I can make a fix for that and you can test it too. Thanks!

raffaeler, 2022-01-26:
@keirf 
> I think anyway it may fail to verify because the verify is not adjusting 360rpm to 300rpm

I could see a clear improvement setting 300 rpm on the command line. I believe the problem is different.
Anyway I'll do the test in a minute. 

FYI I also have a 360K only drive (Mitsubishi MF501B-312M). I can also test it as well, if you tell me there are no specific settings I should set.

keirf, 2022-01-26:
I'm talking about "gw write" which may not be a command you're interested in, and fair enough if so.

raffaeler, 2022-01-26:
@keirf 
Looks like the `--rpm=300` is not recognized when writing:
```
gw.exe write --format ibm.360 --rpm=300 --tracks=step=2 --retries=10 "C:\_floppy\0.img"
*** TEST/PRE-RELEASE: commit 7c5f825ec8180416aab35d536fa63650ca43b397
*** Use these tools ONLY for test and development!!
usage: gw.exe write [options] file
gw.exe write: error: unrecognized arguments: --rpm=300
```

keirf, 2022-01-26:
Yes that's the point. Remove the `--rpm` and see what happens.

Probably the fix is for codecs to adjust the RPM themselves, automatically, both for read and for verify. But let's see.

keirf, 2022-01-26:
Re your 360k drive, you won't need to set pin 2, nor adjust rpm, nor double step. So yes the command line will be simpler.

raffaeler, 2022-01-26:
@keirf 
> Yes that's the point. Remove the --rpm and see what happens.

Done, flawless write

> Re your 360k drive, you won't need to set pin 2, nor adjust rpm, nor double step. So yes the command line will be simpler.

got it

> Probably the fix is for codecs to adjust the RPM themselves, automatically, both for read and for verify. But let's see.

not sure what you mean. I wrote a blank disk and read it back without errors. Should I see an improvement on others? How?

keirf, 2022-01-26:
@raffaeler Actually I was surprised the write, of a 360kB IMG file to your 360rpm drive, worked. :) Since a read without `--rpm 300` doesn't work. Perhaps the bug doesn't exist after all....

keirf, 2022-01-26:
Ah the track speed gets adjusted already for verify. That explains it. Still would be nice for the track decoders to do it automatically. I will investigate that.

keirf, 2022-01-26:
> I was still with my head in the other issue :)
> 
> > @raffaeler Are you still having problems with corrupt IMG being generated?
> 
> Yes, with 7-zip I can't open all the images. I tried PeaZip and it can open all of them. I didn't open the new issue yet because I was trying to understand better the problem

Perhaps 7-Zip is picky about these old FAT filesystems?

raffaeler, 2022-01-26:
> Re your 360k drive, you won't need to set pin 2, nor adjust rpm, nor double step. So yes the command line will be simpler.

FYI I tried with the 360K drive but I get plenty of errors. Not sure whethere the drive is good anyway.

raffaeler, 2022-01-26:
> Perhaps 7-Zip is picky about these old FAT filesystems?

Apparently yes, I will stick to PeaZip.

keirf, 2022-01-26:
Could you give this build a try with your 360 rpm drive and 360kB disks, read and write. You should *not* need the `--rpm 300` option any more. Artifact download link is at bottom of the linked page:
https://github.com/keirf/greaseweazle/actions/runs/1751874292

raffaeler, 2022-01-26:
Done, it works :)

I used the same disk in order to compare the reads, and I read the same disk twice.

First read:
```
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ........................................
0. 2: ........................................
0. 3: ........................................
0. 4: ........................................
0. 5: ........................................
0. 6: ........................................
0. 7: ........................................
0. 8: ........................................
1. 0: X.......................................
1. 1: X.......................................
1. 2: ........................................
1. 3: ........................................
1. 4: ........................................
1. 5: ........................................
1. 6: .....................................X..
1. 7: .....................................X..
1. 8: .....................................X..
```

Second read:
```
H. S: 0123456789012345678901234567890123456789
0. 0: ........................................
0. 1: ........................................
0. 2: ........................................
0. 3: ........................................
0. 4: ........................................
0. 5: ........................................
0. 6: ........................................
0. 7: ........................................
0. 8: ........................................
1. 0: X.......................................
1. 1: X.......................................
1. 2: ........................................
1. 3: ........................................
1. 4: ........................................
1. 5: ........................................
1. 6: ........................................
1. 7: ........................................
1. 8: ........................................
```

After this I wrote the image on a blank disk: no errors.
Finally, I read the written disk back: no errors.

commit 233034f3 (2022-01-27): gw read/write: All format decoders adjust raw data to correct track speed (RPM).

When reads are not index-synced (gw read, gw write verify) use a measurement
we took previously, for flux.ticks_per_rev (fixes crash).

Fixes #153
closed 2022-01-27 by keirf