keirf/greaseweazle #154: Suggestion for retry

raffaeler, 2022-01-25:
I suggest to resync the heading position every `x` tries (to be passed as parameter).
I suspect that the disks presenting bad sectors are not read because of the heading position rather than the floppy itself.

BTW the default PC implementation does this: move back the head to zero, than back to the position and retries.

keirf, 2022-01-25:
What's a good default? Perhaps three reads then reseek, three times total?

raffaeler, 2022-01-25:
Sounds reasonable.
I would advise to parametrize it so that you can try recover the sector for a longer time.

I've seen that re-reading twice in a row the same disk improved the recovery. Probably because the sum of small movements of the head does not match an absolute move (old drive or approximations, who knows).

closed 2022-01-26 by keirf

raffaeler, 2022-01-26:
Super fast, thanks!
@keirf Could you please create a release or link a CI?

keirf, 2022-01-26:
> Super fast, thanks! @keirf Could you please create a release or link a CI?

You can use the latest successful CI build under the repository's Actions tab on its Github page. Click Actions, then click top build, then artifact link is at the bottom of the page.

raffaeler, 2022-01-26:
@keirf I tried the retry version (latest on CI) but I can't see or hear the movement back to sector 0.
Instead, I clearly hear the sound of the track when moving to the next sector.
Could you please check?

commit 62bf6a3f (2022-01-26): gw read: Re-seek after N retries.
Fixes #154