keirf/greaseweazle #155: Suggestion: save log on disk when one or more reads failed

raffaeler, 2022-01-26:
I prefer to save the log on a file to remember which tracks presented the problem.
I tried to redirect gw `stdout` to a file (`> mylog.txt`) but it doesn't get captured.
Also, redirecting is not practical on Windows because I could not see the progress.

The suggestion is to add a parameter that saves the whole log (same of the console) to a file whose name is the same of the image with the `.log` or `.txt` extension.

keirf, 2022-01-26:
Logging goes to stderr.

In Powershell at least you should be able to do something shell-y, like: `gw .... 2>&1 | tee mylog.txt`.

Proper logging would be nicer.

raffaeler, 2022-01-26:
Ah ok, I didn't expect non-errors being sent to `stderr`, in fact PS shows everything in red

There could possibly be two files, one for the info log and the other for the error log. Personally, I would like to only save the one with errors so that I can examine later the results of the reads.

keirf, 2022-01-26:
I think a logging system would allow priority levels to be applied and errors could be flagged as such in the one single log file, for easy filtering. Use of stderr is not super unusual, it leaves stdout free for piping disk images (not actually supported now, but may be in future), and it had a benefit that I could turn off buffering forthose who are piping stderr (GUI wrappers).

raffaeler, 2022-01-26:
Got it, it makes sense.
My scenario is simple: say you have a lot of disks to read: you process one after the other disregarding the console output.
At the end of this batch job, you just take a look at the logs containing errors and validate whether the error may have affected or not the files.

It would be super-useful to know which files are affected from the errors. You already have the filesystem from the --format parameter. Is this complicated?

keirf, 2022-01-26:
I don't do filesystem analysis in Greaseweazle least currently. Adding FAT12 wouldn't be hard I suppose.

raffaeler, 2022-01-26:
> I don't do filesystem analysis in Greaseweazle currently. Adding FAT12 wouldn't be hard I suppose.

If you want to let somebody do it externally, it would be very useful to log the info in a structured way (json, csv, xml, ...) so that I can load the info and the image from my code and provide a feedback on the files being subject to the corruption.

korpx, 2022-09-19:
+1 On logging the result

JPT77, 2025-04-12:
+1 me too