keirf/greaseweazle #157: gw: cannot import name 'version' on RPI/aarch64

midwan, 2022-01-29:
I tested this in two different distros, so far.
Hardware is a Pi400, GW and floppy drive tested on WIndows and WinUAE to be working normally.

### RPIOS-64-bit (https://downloads.raspberrypi.org/raspios_full_arm64/images/)
- Installed software using the steps in the Wiki, no errors reported.
- Connected GW device to USB port.
- Running `gw info` or just `gw`, reports the following traceback:
```
Traceback (most recent call last):
  File "/home/pi/projects/greaseweazle/./gw", line 11, in <module>
    import gw
  File "/home/pi/projects/greaseweazle/scripts/gw.py", line 20, in <module>
    from greaseweazle import version
ImportError: cannot import name 'version' from 'greaseweazle' (/home/pi/projects/greaseweazle/scripts/greaseweazle/__init__.py)
```

### Manjaro (https://www.manjaro.org/downloads/arm/raspberry-pi-4/arm8-raspberry-pi-4-xfce/)
- Installed software using the steps in the Wiki, got some warnings about `wheel` missing and using older method to compile optimized, but otherwise seemed to complete.
- Connected GW device to USB port.
- - Running `gw info` or just `gw`, reports the following traceback:
```
Traceback (most recent call last):
  File "/home/pi/projects/greaseweazle/./gw", line 11, in <module>
    import gw
  File "/home/pi/projects/greaseweazle/scripts/gw.py", line 20, in <module>
    from greaseweazle import version
ImportError: cannot import name 'version' from 'greaseweazle' (/home/pi/projects/greaseweazle/scripts/greaseweazle/__init__.py)
```

Is this a known issue?

keirf, 2022-01-29:
How did you download the Host Tools? You should be downloading the latest official release via: https://github.com/keirf/Greaseweazle/wiki/Download-Host-Tools

If you cloned the repository, what you see is the result. I have just committed a fix to give you a better error message, and to tell you the fix (run "make"). However, downloading a proper versioned release via the wiki is almost certainly what you want to do.

PS. If you still get warnings about "older method used to compile optimized" when you use the latest release, please post the full error message / log snippet here for me to take a look. I may need to update the script to install "wheel". You can try this yourself by adding wheel after requests in scripts/setup.sh

midwan, 2022-01-29:
I cloned the repo, in all cases. :)
I'll check with the zip releases instead, and report back about the missing `wheel` part on the Manjaro distro. Thanks!

midwan, 2022-01-29:
Here's the output from Manjaro (64-bit):

```
[midwan@pi400 greaseweazle]$ sudo ./scripts/setup.sh
Requirement already satisfied: bitarray in /root/.local/lib/python3.10/site-packages (2.3.5)
Requirement already satisfied: crcmod in /root/.local/lib/python3.10/site-packages (1.7)
Requirement already satisfied: pyserial in /root/.local/lib/python3.10/site-packages (3.5)
Requirement already satisfied: requests in /usr/lib/python3.10/site-packages (2.26.0)
Requirement already satisfied: chardet>=3.0.2 in /usr/lib/python3.10/site-packages (from requests) (4.0.0)
Requirement already satisfied: idna>=2.5 in /usr/lib/python3.10/site-packages (from requests) (3.3)
Requirement already satisfied: urllib3>=1.21.1 in /usr/lib/python3.10/site-packages (from requests) (1.26.7)
Processing ./scripts/c_ext
Using legacy 'setup.py install' for optimised, since package 'wheel' is not installed.
Installing collected packages: optimised
    Running setup.py install for optimised ... done
Successfully installed optimised-0.0.0
```

keirf, 2022-01-29:
Please run `python3 -m pip install --user wheel` and try the setup script again

midwan, 2022-01-29:
And here's the output after adding `wheel` in the script, as you suggested:
```
[midwan@pi400 greaseweazle]$ ./scripts/setup.sh
Requirement already satisfied: bitarray in /home/midwan/.local/lib/python3.10/site-packages (2.3.5)
Requirement already satisfied: crcmod in /home/midwan/.local/lib/python3.10/site-packages (1.7)
Requirement already satisfied: pyserial in /home/midwan/.local/lib/python3.10/site-packages (3.5)
Requirement already satisfied: requests in /usr/lib/python3.10/site-packages (2.26.0)
Collecting wheel
  Downloading wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Requirement already satisfied: chardet>=3.0.2 in /usr/lib/python3.10/site-packages (from requests) (4.0.0)
Requirement already satisfied: idna>=2.5 in /usr/lib/python3.10/site-packages (from requests) (3.3)
Requirement already satisfied: urllib3>=1.21.1 in /usr/lib/python3.10/site-packages (from requests) (1.26.7)
Installing collected packages: wheel
  WARNING: The script wheel is installed in '/home/midwan/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed wheel-0.37.1
Processing ./scripts/c_ext
Building wheels for collected packages: optimised
  Building wheel for optimised (setup.py) ... done
  Created wheel for optimised: filename=optimised-0.0.0-cp310-cp310-linux_aarch64.whl size=15159 sha256=8acaf086007b87d617e5cfdbe347d0632bb5078d0fc4ab75b90dcf18dd9f8dab
  Stored in directory: /tmp/pip-ephem-wheel-cache-amktl201/wheels/0c/a8/c4/f7df95f3485902699b3c312272adaf85588acbbb0f9270ae0e
Successfully built optimised
Installing collected packages: optimised
Successfully installed optimised-0.0.0
```

keirf, 2022-01-29:
That looks good!

midwan, 2022-01-29:
It works normally after that, I can call `gw info` and get the expected results, after I disconnected and reconnected the device.
Thanks for the help! :)

closed 2022-01-29 by midwan