keirf/greaseweazle #158: Arch/Manjaro OS: "gw info" reports Not Found

midwan, 2022-01-31:
I've been testing the GW on 3 different Linux ARM distros so far:
- RPIOS 32-bit
- RPIOS 64-bit
- Manjaro OS 64-bit (based on Arch)

I've installed the host tools from the repo, in all 3 scenarios:
`make && scripts/setup.sh`

Things work perfectly on RPIOS 32-bit. `gw info` reports the device normally, and it works fine in Amiberry as well (with the `FloppyBridge.so` library by Rob Smith).

On RPIOS 64-bit, `gw info` still reports the device normally, but it doesn't work on Amiberry. I suspect this is related to the `FloppyBridge.so` part however, as I can read a disk to an image using `gw`.

But I have a strange situation on Manjaro OS, where `gw info` reports that the device was `Not Found` always.
The device is connected and detected by the OS, from what I can see. Here's an extract from `dmesg`:

```
[ 5550.499785] usb 1-1.2: new full-speed USB device number 5 using xhci_hcd
[ 5550.607137] usb 1-1.2: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
[ 5550.607145] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5550.607150] usb 1-1.2: Product: Greaseweazle
[ 5550.607153] usb 1-1.2: Manufacturer: Keir Fraser
[ 5550.607157] usb 1-1.2: SerialNumber: GW135195064205400002741705
[ 5551.238845] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[ 5551.241123] usbcore: registered new interface driver cdc_acm
[ 5551.241132] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

And yet:
```
[midwan@pi400 greaseweazle]$ ./gw info
Host Tools: v0.38
Device:
  Not found
```

From what I could see in the Python sources, this would happen if an exception was thrown while trying to open the serial device. Is there a way to get more detailed output, to try and figure out what might be going on here?

keirf, 2022-02-01:
Run `gw --bt info` to get a backtrace. You will additionally want to remove the `try/except` wrapper around `util.usb_open()` in `tools/info.py`, or the exception gets consumed.

One possibility here might be that a daemon or driver has ttyACM0 open, or has written garbage to it (eg. modem probing). There is a udev script `49-greaseweazle.rules` which although tested on Ubuntu likely applies as-is to Arch/Manjaro too.

keirf, 2022-02-01:
That looks like a pretty hard error on opening `/dev/ttyACM0`. Can you open it using any other tools eg `miniterm`?

midwan, 2022-02-02:
@keirf 
I've tried multiple things, even a different kernel, but in all scenarios I cannot access `/dev/ttyACM0` no matter what I do 😞 
It's very strange, and I've run out of ideas to try for now. I'll let you know if I make any progress...

keirf, 2022-02-02:
Do you see anything interesting in dmesg on failure to open?

midwan, 2022-02-02:
Nothing unexpected, from what I can see:
```
[  194.183851] usb 1-1.2: reset full-speed USB device number 6 using xhci_hcd
[  194.289921] usb 1-1.2: USB disconnect, device number 6
[  194.371586] usb 1-1.2: new full-speed USB device number 7 using xhci_hcd
[  194.478861] usb 1-1.2: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
[  194.478872] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  194.478876] usb 1-1.2: Product: Greaseweazle
[  194.478880] usb 1-1.2: Manufacturer: Keir Fraser
[  194.478883] usb 1-1.2: SerialNumber: GW135195064205400002741705
[  194.481522] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
```

I did another test on a fresh Debian 64-bit Bookworm, which also worked fine. This seems to be a problem only in Manjaro, but I have no idea why.

keirf, 2022-02-02:
Perhaps the device is open already? Does `sudo lsof /dev/ttyACM0` show anything?

midwan, 2022-02-02:
```
[midwan@pi400 greaseweazle]$ sudo lsof /dev/ttyACM0
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
```

I also checked if there was any conflict with ModemManager, but that service is not loaded anyway:
```
systemctl stop ModemManager.service
Failed to stop ModemManager.service: Unit ModemManager.service not loaded.
```

I know that Manjaro/Arch is a bit more restrictive with permissions, since when I was testing with the Arduino controller (Rob Smith's one) I couldn't access the tty device until I was a member of the group. So I checked and added my user to all relevant groups as well: `uucp`, `tty` are included.

This is how the device looks like:
```
[midwan@pi400 greaseweazle]$ ls -al /dev/ttyACM0
crw-rw-rw- 1 root uucp 166, 0  2 feb 16.54 /dev/ttyACM0
```

I tried using `screen` to connect to it, which works on RPIOS (where the device is detected from `gw info`) but fails on Manjaro immediately:
```
[midwan@pi400 greaseweazle]$ screen /dev/ttyACM0
[screen is terminating]
[screen is terminating]
```

keirf, 2022-02-02:
Could be a kernel bug/issue? https://forum.manjaro.org/t/usb-device-dev-ttyacm0-throws-i-o-error-on-connect/61385

midwan, 2022-02-02:
That looks very possible, thanks :) I'll wait until they push the update and test it again there.

andrewbird, 2022-02-02:
> So I checked and added my user to all relevant groups as well: uucp, tty are included.

After you added yourself to those groups did you log out and back in to collect the new permissions? You can always check what you have at present with `id -a`

midwan, 2022-02-02:
@andrewbird 
Yes, I did. :)
I swapped SDcards on the same device during tests (to try the same on other distros), so the whole system was shutdown and started up between them.

closed 2022-03-22 by keirf