keirf/greaseweazle #159: greaseweazle.usb.CmdError

Venomtek, 2022-02-03:
When I try to read or write "track 0" I get this error.  If I start at track 1, no error.  Only happens with this particular drive.
Chinon FZ-502 (5.25 360K)

----
PS D:\emu\Greaseweazle\greaseweazle-tools-v0.38> .\gw.exe write --tracks c=0-35:h=0 ..\BOOT.SCP
Writing c=0-35:h=0
Traceback (most recent call last):
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\tools\write.py", line 221, in main
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\tools\util.py", line 253, in with_drive_selected
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\tools\write.py", line 51, in write_from_image
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\usb.py", line 229, in seek
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\usb.py", line 216, in _send_cmd
greaseweazle.usb.CmdError: <exception str() failed>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\__startup__.py", line 113, in run
  File "C:\Users\runneradmin\AppData\Roaming\Python\Python38\site-packages\cx_Freeze\initscripts\Console.py", line 15, in run
  File "gw.py", line 99, in <module>
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\tools\write.py", line 223, in main
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\usb.py", line 152, in __str__
  File "D:\a\greaseweazle\greaseweazle\greaseweazle-tools-v0.38\scripts\greaseweazle\usb.py", line 148, in errcode_str
struct.error: unpack requires a buffer of 3 bytes
PS D:\emu\Greaseweazle\greaseweazle-tools-v0.38>
----

keirf, 2022-02-03:
Thanks you, that's definitely a bug in an error path in the Python code. I will fix that.

However, you shouldn't really be in that error path in the first place. I think this drive isn't seeking reliably: Greaseweazle expects the drive to be at cylinder 0, but it appears not to be. Possibly you need to increase the "step delay" to increase seek times. Try this before any other drive commands:
```
gw delays --step 20000 --settle 30
```

Venomtek, 2022-02-03:
> Thanks, that's definitely a bug in an error path in the Python code. I will fix that.
> 
> However, you shouldn't really be in that error path in the first place. I think this drive isn't seeking reliably: Greaseweazle expects the drive to be at cylinder 0, but it appears not to be. Possibly you need to increase the "step delay" to increase seek times. Try this before any other drive commands:
> 
> ```
> gw delays --step 20000 --settle 30
> ```

Thanks but this drive has major issues.  I've just never had this error before no matter how bad a drive is so I thought I'd let ya know.

Thanks for all your hard work!

closed 2022-02-04 by keirf