keirf/greaseweazle #165: RX02 Format Support

mkatzNU, 2022-02-03:
The Digital Equipment Corporation RX02 drive uses the standard IBM 3740 format (77 tracks, 26 sectors) but the sector data is double density (256 bytes instead of 128) while the sector header is single density.

Add the ability to switch density between the sector header and the sector data.

keirf, 2022-02-04:
RX02 is understood by my companion GitHub project disk-utilities. So you can convert between sector images and (say) scp flux format and should be able to read/write the latter to eight inch disks using Greaseweazle. GW does not natively support the format currently but it could for sure be added.

keirf, 2022-02-04:
A good first step would be to see if a workflow incorporating disk-utilities actually works. If you can get a test setup working in this way, it can be used for testing RX02 features added directly to gw.

mkatzNU, 2022-02-04:
I ordered my Greaseweazle from Tindie as well as the 34 to 50 pin 
adapter today.

Unfortunately that is only one small part of the entire project. When I 
am fully up and running I will let you know.

Here are the tasks to my project:

1.  Order Greaseweazle & Adapter
2.  Order/Build 24V & 5V supplies
3.  Order connectors for DC and AC power supplies
4.  Connect Greaseweazle to drive for initial testing (IBM 3740 format)
5.  Install RX01/RX02 interface into PDP-8/E
6.  Build OS/8 with RX01 driver in it.
7.  Test Greaseweazle IBM 3740 format
8.  Build OS/8 with RX02 driver in it.
9.  Test Greastweazle RX02 format.

I am several weeks or more away from being ready to test new code.

Thank you for your time,

                Mike

-- 

--

*Mike Katz*

*Sr. Software Architect*

www.nucurrent.com <http://www.nucurrent.com/>
641 W. Lake St., Suite 304
Chicago, IL 60661

(o) +1 (312) 637-3511
(c) +1 (773) 414-1044

The preceding email message may be confidential or contain protected 
information. Neither the message nor the content herein are intended for 
transmission to, or receipt by, any unauthorized persons. If you have 
received this message in error, please (i) do not read it, (ii) reply to 
the sender that you received the message in error, and (iii) erase or 
destroy the message.

keirf, 2022-02-04:
That's fine, this ticket can serve to track progress.

mkatzNU, 2022-08-07:
I finally have the Greaseweazle up and running on my SA-800 drive.

I am able to get my SA-800 to seek.  I haven't tried reading or writing yet.

I can't find full documentation for the Greaseweazle.  The Github Wiki is a little sparse.  For example there is no description of the known format types (produced with gw read -h).  I know the firmware can handle the IBM 3740 format of the RX01 but I didn't see anything resembling that in the format list.

Is there any way, in Greaseweazle, to create a drive profile so that I can refer to the profile for all commands.

The SA-800 is:

Heads:    1
Tracks:    77
Seek:       8mS
Density:  Single Density (RX01) and Double Density (RX02 Sector Data only)

One interesting note on the gw firmware.  When 'I tried to run a seek test without specifying a drive, I got a track zero cannot be found error.  However, when I did the same thing with the clean command it drive selected automatically.  I don't know if clean selects all drives or this is a bug.

My next step is to read a diskette.  Do you have any suggestions?

As always,

        Thank you.

Thanks,

            Mike

keirf, 2022-09-03:
You should be able to access the drive probably as drive 0. Try `gw seek --drive=0 40` for example.

keirf, 2023-04-24:
Any update to this before I scrub this ticket?

closed 2023-05-10 by keirf