keirf/greaseweazle #168: Floppy drive LED always on, even when not in use

ssterling, 2022-02-15:
Each connected floppy drive’s in-use indicator LED is always on, even when the Greaseweazle isn’t reading anything.  Hooking the same drives up to a regular controller, the LED acts as expected.  It’s just a mild nuisance more than anything, but I still thought I’d file an issue.

keirf, 2022-02-16:
Not heard of that before. What type of Greaseweazle? And drives?

ssterling, 2022-02-16:
I’m running a V4.  The drives are a Chinon FR-506 RevB and an NEC FD1231H.

andrewbird, 2022-02-16:
I'm not sure this is your answer, but a long time ago I experienced something similar with a std pc when I connected the floppy cable upsidedown. Are you sure of your connections?

ssterling, 2022-02-17:
And here I thought the Greaseweazle itself not working was just me not knowing how to use the tools on Linux.  Yup, notchless connector was inserted upside-down LOL

closed 2022-02-17 by ssterling