keirf/greaseweazle #178: MapleMini port

d0plr, 2022-03-05:
Hi,

I've been trying to adapt the GW firmware to the Maple Mini clone board from Baite, but something eludes me. The pin mapping is slightly different from the BluePill: http://docs.leaflabs.com/static.leaflabs.com/pub/leaflabs/maple-docs/latest/hardware/maple-mini.html . See initial patch attached.  Here are the symptoms:

- With the gw firmware, the led turns on but nothing is detected on the usb host. The stlink adapter then fails to detect the maple mini altogether. Using a serial adapter to write back the original bootloader works though.

- With the blinky firmware, the led does not turn on at all, but the output on the serial output looks ok (see below). I did change the led gpio from PC13 to PB1 (PB1 works fine with arduino sketches).

```
** Blinky Test **                                                   
** Keir Fraser <keir.xen@gmail.com>                                 
** https://github.com/keirf/Greaseweazle
Serial = ff48:066e:8749:5149:5821:8716
Flash Size = 128kB (STM32F103xB Medium-Density)
Device ID = 0x0000
Revision  = 0x0000
Testing I2C1... OK
Testing I2C2... OK
Testing SPI1... OK
Testing SPI2... OK
Testing TIM1... OK
Testing TIM2... OK
Testing TIM3... OK                                                              
Testing TIM4... OK                                                              
DMA Test #1... OK                                                               
DMA Test #2... OK                                                               
DMA Test #3... OK                                                               
DMA Test #4... OK                                                               
Testing 128kB Flash... OK                                                       
Enable TIM4 IRQ... .OK                                                          
Testing 20kB SRAM (endless loop).....................
```

I tried 2 maple mini boards bought years appart, same result. Note: in case that matters I did not change the pullup on USB D+ (I don't actually know how to do that on this board).

Can you see what I'm missing ?
If you're interested to have this included in your repo, I could rewrite it to keep both BluePill and MapleMini compatibility (the easiest way would be to duplicate the stm32f1 folders as if this was a different mcu I guess).

[gw-maplemini-diff.txt](https://github.com/keirf/greaseweazle/files/8190311/gw-maplemini-diff.txt)

Thanks

keirf, 2022-03-06:
Port pin PB9 is used to connect/disconnect the pullup on the USB D+ line. You'll need to configure PB9 as an output and drive it either low or high (I can't immediately tell which) for USB to be detected by the host.

See the schematic: https://github.com/leaflabs/maplemini/blob/master/maplemini.pdf

keirf, 2022-03-07:
Yes, mysterious. The LED ought to blink. Maybe printk a character in the TIM4 IRQ handler and see if that's actually getting called.

d0plr, 2022-03-07:
There is already a printk(".") in there, and I can see it in the serial output.

d0plr, 2022-03-07:
OK got it. I have to configure PB! to push/pull instead of open drain. One mystery solved.

d0plr, 2022-03-13:
PB9 (USB_CON) had to be pulled down instead of up, now the adapter responds to USB.

closed 2022-03-22 by keirf