keirf/greaseweazle #179: Greaseweazle "the brick"

Pit50, 2022-03-09:
Hello!
I have a problem with Greaseweazle V4.
At first it was ok, I read data from one floppy disk, and suddenly Greaseweazle stopped working.  Windows doesn't recognize the device, it displays it as AT32 Bootloader DFU.  What happened?  What can I do with this?  I tried to install the firmware with ArteryISP but it didn't help.  What should I do?  Help me please!

keirf, 2022-03-09:
The board will enter DFU update mode if the DFU pin is jumpered to 3V3 (see the left-most two pins on the 10-pin header). If a jumper is not in that position then perhaps there is a solder blob making partial contact underneath the jumper block? In any case, if it is not explicitly jumpered for DFU mode as described above, this can be considered a board fault and you should seek a replacement.

Pit50, 2022-03-09:
Thank you for your answer. 
I bought a Greaseweazle a few days ago at RetroShop in Portugal and already wrote to them. They agreed to replace the device.
Greetings.
Piter

closed 2022-03-09 by keirf