keirf/greaseweazle #181: Request: 2.88M "ED" Support

Compgeke, 2022-03-21:
I may just be missing the options, but I was wondering if there's (planned) support for 2.88M "ED" image support. I've got some disks here in that format from NeXT gear, IBM gear and a Nortel SL-1 as well as a couple regular 34-pin ED drives. 

If any testing is needed for this, I can do it - I'm aware finding actual 2.88M drives outside an IBM PS/2 is incredibly difficult.

keirf, 2022-03-21:
You will likely find that you get flux overflow errors trying to read ED disks, unless you happen to have an F7 Lightning Plus device. This is because ED requires available USB bandwidth around 8Mbps, which is pushing it over USB Full Speed. Whereas the F7L+ device has USB High Speed.

This could probably be worked around in a few ways. Eg. decode/encode MFM on the Greaseweazle device. Or smarter encoding of flux (ie. include some compression).

markusC64, 2022-03-22:
Well, I can confirm that I have read a full C64 CMD FD4000 Disk (that is ED) with m yGreaseWeazle V4 without any flux overflows. Tested with Mitsubishi MF356F-252MG.
That CMD Floppy uses standard MFM ED Bitrate.

BTW: You can use kryoflux dtc.exe for decoding flux data. dtc does not officially support ED density for decoding, but with additional parameter "v150" decoding works fine. That way, I am able to decode the flux image from the CMD FD4k disk.

keirf, 2022-03-22:
It's definitely marginal. If you get 8mbps minimum read bandwidth in gw info then that's enough. Just.

markusC64, 2022-03-22:
For sure. Anyway, it's worth a try.

BTW: Isn't there small buffering? In that case, only the average data rate must fit. And MFM disks do not have to many consecuting ones.

keirf, 2022-03-22:
The marginality already accounts for the lack of consecutive ones. Data rate is 1MHz (that's the unencoded non mfm rate). Each flux sample is a byte. So that's 8mbps.

keirf, 2022-03-22:
There is buffering. But gw info already accounts for it in its minimum rate. After all rate has to be measured across *some* time. So time to fill the buffer is what's used.

hadess, 2022-05-24:
Could bandwidth also be a problem for "DMF" floppies as used by Microsoft? They're listed as "1680 KB" and "1720 KB". If that doesn't butt against a bus bandwidth limit, I'll file a separate issue about it.

markusC64, 2022-05-24:
It shouldn't. They do not have a higher data rate than standard disks. Only shorter gaps between the sectors.

keirf, 2022-06-25:
@Compgeke are you still interested in testing this?