keirf/greaseweazle #186: Flux Visualization

jepler, 2022-04-07:
Hi! Is this a sort of feature you'd like to see contributed to greaseweazle, or is it better kept separate?  I ask because using greaseweazle code to read .scp files helped provide a short path to producing this visualization.

If you're interested, I'll work to integrate the code in gw via pull request.  It would add a couple of big new prerequisites, numpy and skimage.

[image]

jepler, 2022-04-07:
The current code lives at https://github.com/jepler/fluxvis/ and can be licensed as needed.

ssterling, 2022-04-13:
My $0.02: integrating this *into* Greaseweazle host tools would sorta bloat the codebase past its purpose of performing low-level operations on floppy drives.  However, adding, say, a `--visualize` option to the `read` module—dependent on whether fluxvis is found in `$PATH`—could invoke it with the parameters that were already passed to Greaseweazle.  Please take what I say with a grain of salt, though; I actually know nothing about ideal software design LOL

jepler, 2022-04-13:
We were inspired by the flux visualization of applesauce, which people seem to like. It can also help mere humans understand floppy disk structure and problems, such as spiral protection tracks, "degaussed" and dirty media. https://wiki.reactivemicro.com/Applesauce#Applesauce_Image_File_Format_.28.WOZ_and_.A2R.29

It would be interesting to combine the flux density visualization with flux interpretation, possibly using fluxengine's csv format: http://cowlark.com/fluxengine/doc/using.html#visualisation

[image]

jepler, 2022-04-13:
But as we have the word of the maintainer that this is best NOT added to gw, happy to close this up and let fluxvis live on its own! thanks for your time, as well as for the GW code for reading flux files which made it possible to write this tool.

closed 2022-04-13 by jepler

keirf, 2022-04-13:
To be direct: I'll be happy to see visualisation in greaseweazle but to be useful it needs flux interpretation and that is bound to need closer integration with the greaseweazle core and indeed changes to the greaseweazle core. 1s and 0s, or flux length interpretation, is pretty but unenlightening. Thank you for the request and starting a discussion!