keirf/greaseweazle #187: Rephrasing the Blue Pill R10 soldering fix

shinedeveloper, 2022-04-08:
Hi,

I suggest to rephrase the R10 fix instructions on page ( https://github.com/keirf/greaseweazle/wiki/Blue-Pill-Adapter ). It currently does not mention 'why' one should do it thus confusing me. I had to look it up on the internet first to know whether or not to apply the fix.

My suggested text:
There are some Blue Pill boards (non-original clones) on the market that have incorrect R10 resistor value 10 kΩ instead of the correct 1.5 kΩ. If (and only if) it applies to your Blue Pill board, apply the fix by placing the 1.5 kΩ resistor at the location marked **USBPU** and removing the R10 from the Blue Pill. You **must** remove R10 if you ...

keirf, 2022-04-08:
It also means the USB connection is correctly reset if the Blue Pill is reset, but I suppose that's not very useful to most users. It can also allow jumperless update, but since most Blue Pill users don't have the USBPU mod I'm unable to rely on it and hence jumpered update is mandated for basic F1 models.

keirf, 2022-04-08:
I might simply say to ignore **USBPU** actually. The boards with the 10k R10 were sold long ago.

closed 2022-04-08 by keirf