keirf/greaseweazle #190: Always sectors missing, when reading 5,25" IBM disks

explit7, 2022-04-17:
Hello keirf, i am quite new with your great hardware. Trying to read my old IBM DOS 5,25" Floppy disks. From 10 disks i was able to read only 1 without any problems. the others shows me always, that sectors are missing...

maybe i do something wrong, ore all my disks are all broken?

[image]

ssterling, 2022-04-19:
I got some similar garbage until I [cleaned my drive head](https://www.retrotechnology.com/herbs_stuff/clean_disks.html#heads).  Turns out the 3½″ drive I salvaged was *dusty*.

keirf, 2022-04-19:
Another possibility is density select on floppy pin 2 eg `gw pin set 2 L`

explit7, 2022-04-19:
Thanks for the answers guys. Already purchased a cleaning disk on ebay.
@keirf My Floppy is Mitsumi D509V2.
[image]

Can i change the density with correct Jumper settings ?

keirf, 2022-04-19:
I don't know but usually the drive is told the density by the host, via signalling on pin 2. Also the HD drive usually spins at 360rpm regardless of density (DD media is notionally accessed at 300rpm) but FlashFloppy should automatically fix that up these days.

Hence the suggestion to set pin 2 Low via `gw pin set 2 L`. And then retry your read.

explit7, 2022-04-19:
gw pin set 2 L unfortunately = the same result. Looks like the drive head is really dirty :(

keirf, 2022-04-19:
You can also try adding `--rpm 300` to the `gw read` command line.

explit7, 2022-04-19:
--rpm 300 is not recognized as command unfortunately, also rpm=300 is not recognized....hm..

keirf, 2022-04-19:
Sorry the parameter now is `--adjust-speed 300`

keirf, 2022-04-19:
I see that early on you say the one disk at least did read correctly. Does it still read correctly? As well as being themselves unreadable, really dirty disks can also dirty or even damage your drive heads.

explit7, 2022-04-19:
> I see that early on you say the one disk at least did read correctly. Does it still read correctly? As well as being themselves unreadable, really dirty disks can also dirty or even damage your drive heads.

I was able to read a disk with a very odd format. It was 5,25" disk with 720KB. (really odd), but i could read it correctly. afterwards i tried multiple times to read the same and other disks - always with the same error.
So i think, the drive is dirty or bad. Need to find another 5,25 Floppy to test with them

Thank you @keirf for your help!

closed 2022-04-26 by keirf