keirf/greaseweazle #192: Sector missing with V4 on 5.25" but not with F1

badre2911, 2022-04-26:
Hi,
I just got the V4 version of greaseweasle, but it doesn't work with my 5.25" YD-380B player, but the bluepill F1 version works fine.
I only have missing sectors with the V4 version.
The floppy is 1.2mb and works well on another 5.25" drive.
Is there a GW parameter for this YD-380D player to put?
Thanks.

keirf, 2022-04-26:
All sectors missing? It should plug and play, same as F1.

keirf, 2022-04-26:
Try dump to SCP and attach and I could have a look and see if there's anything obvious.

badre2911, 2022-04-26:
Yes its dumped with V4.
With GW console i have this
[image with gw](https://imgur.com/74hQggK)
When i open it with hxc i have
[image open with hxc](https://imgur.com/XwianzX)

With GW from FluxMyFluffyFloppy_3.05 (it use gw)
on output I have this until track 9
[image](https://imgur.com/7ufwqCU)
after the tracks 9 all is ok
[image  at end](https://imgur.com/GTqDqpw)
when i open with HxC
[image open with HxC](https://imgur.com/Cy4qIoc)

With F1 i have no errors with GW and FluxMyFluffyFloppy and i can open it  with HxC , but not with V4

keirf, 2022-04-26:
So on console it looks fine? Make sure gui is not adding an --rpm parameter. You will need to ask the gui author for further support on that.

badre2911, 2022-04-26:
on console (image2.scp) it look fine on screen, but i can't open it, i can't see the content of the files present in the image, i think the file is corrupted.

But with image3.scp, this file is not corrupted, i can see the content of files with HxC, but there are errors on screen.

The problem is with gw on console mode, the size file image2.scp is not a same than image3.scp.

badre2911, 2022-04-26:
Now all is ok,
I have put adjust speed to 300 and forced cylinders to 0-79, and head to 0, all work fine.
if i put parameter Heads to 0-1 it don't work.

badre2911, 2022-04-26:
with this parameter on console mode all fork fine now.

`gw read --adjust-speed=300rpm --format ibm.1200 --tracks c=0-79:h=0 image5.scp`

I can close this issue.

closed 2022-04-26 by badre2911