keirf/greaseweazle #193: Return non-zero exit code when "info" fails to find a device

hadess, 2022-05-02:
It would be useful for integration in scripts if `gw` could return a non-zero value from the tool if there aren't any devices detected.
```sh
$ gw info  ; echo $?
*** TEST/PRE-RELEASE: 0.43.dev5+g39b5d2d
*** Use these tools ONLY for test and development!!
Host Tools: 0.43.dev5+g39b5d2d
Device:
  Not found
0
```

`src/greaseweazle/tools/info.py` contains that code:
```
    except serial.SerialException:
        print('  Not found')
        sys.exit(0)
```

hadess, 2022-05-04:
It looks like gw also doesn't return 1 on other failures, such as reading or writing errors. I can see cases where this isn't wanted, but it would be great if there was a way to make such errors fatal and returning an error.

keirf, 2022-05-04:
That's a trickier one. Perhaps a command line option is warranted. Your original suggestion certainly makes sense though.

Albert-S-Briscoe, 2022-08-28:
I just ran into this too. `--exit-codes` or something similar would be great.

For detecting a device, attempting `gw seek 0` will return an error when there is no greaseweazle, but I haven't found a way of detecting when a particular drive is present other than parsing stdout for errors.

JPT77, 2025-04-12:
I agree GW needs exit codes. 

I would like to log the bad sectors map to file if it did not read all sectors.
but turns out the result code is always zero.  :(

Btw, why goes all output to stderr? 
everything should go to stdout. stderr should be empty except there was an error.
should I create a new issue?

keirf, 2025-04-12:
Not unusual as it keeps stdin/out clean for piping in future.

keirf, 2025-04-12:
In fact historically it is more normal to log to stderr than stdout.