keirf/greaseweazle #194: adf write error

guy426, 2022-05-14:
V4 board (firmware versione 1.1) don't write ADF image:
E:\doc\floppy\greaseweazle-0.42>gw write --format amiga.amigados amigatestkit.adf
Writing c=0-79:h=0-1
Format amiga.amigados
T0.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T0.0: Writing Track (Verify Failure: Retry #1)
T0.0: Writing Track (Verify Failure: Retry #2)
T0.0: Writing Track (Verify Failure: Retry #3)
** FATAL ERROR:
Failed to verify Track 0.0

I changed 10 disk, same results.
Supports are good, ibm write image works well on every disk.

keirf, 2022-05-14:
You don't need the --format option if writing an amiga adf. Are the disks HD? If so have you covered the HD hole?

guy426, 2022-05-15:
Some results...
I' have try also with low density disk and with HD hole covers.

keirf, 2022-05-15:
Bad drive failed HD sensor perhaps? If its writing HD disks okay there's otherwise no reason DD shouldn't work. I can assure you it's not a software issue.

guy426, 2022-05-16:
I have replaced my floppy disk reader and now works well. Thanks

closed 2022-05-16 by guy426