keirf/greaseweazle #195: Linux continually disconnects/reconnects the Greaseweazle

nmelnick, 2022-05-15:
Hi! Opening a new ticket as I'm seeing similar symptoms as #120 .

Similar to that ticket, I have a greaseweazle connected to my PC. In Windows 10, it works just fine -- I am able to read and write disks successfully using the provided packages, and also used it to update the firmware to the latest revision. I normally use Linux on this machine, though, and that's where I'm seeing issues.

Machine is a Lenovo ThinkCentre M92p - Intel Core i7-3770, Q77 chipset. Windows side is Windows 10 21H2, Linux side is Fedora 36.

I installed the software using ```python3 -m pip install git+https://github.com/keirf/greaseweazle@latest```. The provided udev rules were installed and reloaded using udevadm. I have uninstalled ModemManager to avoid continued issues, but it changed nothing.  When executing, say, ```gw info```, dmesg will show the device doing a reset and detection event:

```
[  162.926920] usb 3-2: reset full-speed USB device number 7 using xhci_hcd
[  163.054075] usb 3-2: USB disconnect, device number 7
[  163.168710] usb 3-2: new full-speed USB device number 8 using xhci_hcd
[  163.296138] usb 3-2: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
[  163.296147] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  163.296150] usb 3-2: Product: Greaseweazle
[  163.296152] usb 3-2: Manufacturer: Keir Fraser
[  163.296154] usb 3-2: SerialNumber: GW1351951442054000029C1705
[  163.297431] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
```

Executing the command over and over will eventually succeed and provide a result, but a subsequent run will show an error during detection. I haven't tried to brute force a read or write event yet.

I have tried to remove it from the hub and direct connecting it, with no change. I am powering the drive from external power, so another suggestion regarding a USB brownout doesn't seem to apply. The Windows side reads/writes without issue, even with direct connected power.

Happy to provide more information as needed. I can use the device fine in the meantime, would just love to avoid the reboot. :)

Thank you for the excellent work on this device.

keirf, 2022-05-15:
Hi. The previous ticket was inconclusive. Does this happen if you `gw info` even with no drive connected at all; just the gw board connected by itself via USB? I wonder if the reset is triggered by Linux or the device. I'm thinking Linux but I'm not sure why only some setups seem to see this.

VorpalBlade, 2022-06-08:
I had a very similar issue when I first bought my GW. However I quickly realised the USB cable was bad, and after borrowing the cable for the USB hub in my external monitor (the only one with USB A-B I could find) it worked just fine. Checking the cable is worth it.

closed 2022-07-13 by keirf

mikijov, 2023-02-08:
To add, the same issue can happen when connecting through a USB hub.