keirf/greaseweazle #197: Add DMF support

hadess, 2022-05-25:
Based on this [comment](https://github.com/keirf/greaseweazle/issues/181#issuecomment-1136312063), I'm filing this issue for Fluxengine support for DMF files. I'm interested in reading that format to verify some original software I have, but I could easily imagine someone being interested in writing to make backup copies.

I use a Greaseweazle (v4), and have access to MS Works '95 installation disks which use DMF for disk 2 onwards.

Happy to test support, or help fund it. Let me know.

I've also filed an issue against Fluxengine for the same feature:
https://github.com/davidgiven/fluxengine/issues/514

markusC64, 2022-05-25:
Just in case you'll need a dmf image together with its expected content:

[dmf.7z.zip](https://github.com/keirf/greaseweazle/files/8772987/dmf.7z.zip)

hadess, 2022-05-28:
See also: https://en.wikipedia.org/wiki/Distribution_Media_Format

keirf, 2022-05-28:
Sorry I meant to look at this. Support will be trivial to add.

hadess, 2022-05-28:
Some more disk images:
https://torlus.com/floppy/forum/viewtopic.php?p=23790&sid=0966cc635602e864398f2abb07869634#p23790

> Sorry I meant to look at this. Support will be trivial to add.

I tried, but failed. I guess just bumping the number of sectors and scaling the clock wasn't enough, so I'm curious so as to what other changes are needed. Thanks!

keirf, 2022-05-28:
Its like IBM.1440 but more sectors and small/no gap3. The clock doesn't change.

hadess, 2022-05-28:
My attempt:

```patch
diff --git a/src/greaseweazle/codec/formats.py b/src/greaseweazle/codec/formats.py
index 2041c9b..5fa1a5d 100644
--- a/src/greaseweazle/codec/formats.py
+++ b/src/greaseweazle/codec/formats.py
@@ -151,6 +151,14 @@ class Format_IBM_1200(Format):
         self.default_revs = m.default_revs
         super().__init__()
 
+class Format_IBM_DMF(Format):
+    img_compatible = True
+    def __init__(self):
+        import greaseweazle.codec.ibm.mfm as m
+        self.fmt = m.IBM_MFM_DMF
+        self.default_revs = m.default_revs
+        super().__init__()
+
 class Format_Atari_90(Format):
     img_compatible = True
     default_trackset = 'c=0-39:h=0:step=2'
@@ -249,6 +257,7 @@ formats = OrderedDict({
     'ibm.720': Format_IBM_720,
     'ibm.1200': Format_IBM_1200,
     'ibm.1440': Format_IBM_1440,
+    'ibm.dmf': Format_IBM_DMF,
     'sega.sf7000': Format_Sega_SF7000,
 })
 
diff --git a/src/greaseweazle/codec/ibm/mfm.py b/src/greaseweazle/codec/ibm/mfm.py
index 79efe19..855bb93 100644
--- a/src/greaseweazle/codec/ibm/mfm.py
+++ b/src/greaseweazle/codec/ibm/mfm.py
@@ -393,6 +393,12 @@ class IBM_MFM_1440(IBM_MFM_720):
     clock = 1e-6
     nsec   = 18
 
+class IBM_MFM_DMF(IBM_MFM_720):
+
+    clock = 1e-6
+    nsec   = 21
+    gap_3  = 0
+
 class AtariST_SS_9SPT(IBM_MFM_720):
 
     gap_1 = None
```

Tested with `gap_3` set to `0` and `30`. Do I need to test every value of gap_3 until I find the right one, or is there a more "scientific" way to get/calculate this info?

hadess, 2022-05-28:
I'm testing with:
```sh
$ gw read --format ibm.dmf  works1.img
*** TEST/PRE-RELEASE: 0.43.dev5+g39b5d2d.d20220528
*** Use these tools ONLY for test and development!!
Reading c=0-79:h=0-1 revs=2
Format ibm.dmf
T0.0: IBM MFM (18/21 sectors) from Raw Flux (154136 flux in 401.07ms)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (385257 flux in 1002.45ms) (Retry #1.1)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (616369 flux in 1603.81ms) (Retry #1.2)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (847488 flux in 2205.19ms) (Retry #1.3)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (1078607 flux in 2806.56ms) (Retry #2.1)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (1309728 flux in 3407.94ms) (Retry #2.2)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (1540845 flux in 4009.29ms) (Retry #2.3)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (1771960 flux in 4610.65ms) (Retry #3.1)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (2003077 flux in 5212.02ms) (Retry #3.2)
T0.0: IBM MFM (18/21 sectors) from Raw Flux (2234195 flux in 5813.37ms) (Retry #3.3)
T0.0: Giving up: 3 sectors missing
T0.1: IBM MFM (18/21 sectors) from Raw Flux (181303 flux in 401.07ms)
```

keirf, 2022-05-28:
If you're seeing 18 of 21 sectors perhaps this is a normal 1m44 disk? First disk in an install set surely would be.

hadess, 2022-05-28:
> If you're seeing 18 of 21 sectors perhaps this is a normal 1m44 disk? First disk in an install set surely would be.

:facepalm:

Disk 2 ripped properly (`Found 3360 sectors of 3360 (100%)`). Do you want me to create a PR for it?

keirf, 2022-05-28:
It's okay there's a few more parameters I can pull out from Flashfloppy involving skew and interleave. I'll sort a final patch for you to test this afternoon.

hadess, 2022-05-28:
> It's okay there's a few more parameters I can pull out from Flashfloppy involving skew and interleave.

I saw there's also a 2:1 interleave, but couldn't find anything else specific to DMF in flashfloppy.

> I'll sort a final patch for you to test this afternoon.

Awesome, thanks very much.

markusC64, 2022-05-28:
I happen tpo have some .scp images of DMF disk. I'll be happy to test the final patch too when available with that images.

keirf, 2022-05-28:
This is now fixed in the source tree. I have also added reporting of unexpected sectors in a track. So, for example, if you dump a DMF disk with `--format ibm.1440` you will now receive warnings that unexpected sectors are being ignored. Which is handy!

If you want a Windows build you can find it here (Linux/Mac can install from source using pip):
https://github.com/keirf/greaseweazle/suites/6697170287/artifacts/254395682

markusC64, 2022-05-28:
Good news: I decoded the 1st of my dmf kryoflux / scp files with "gw convert". It results in a binary that is equal to the one that dtc.exe decoded.
Converting the dmf img file to flux also looks good, but I don't have a PC with floppy controller to do a real test.

keirf, 2022-05-28:
Sounds good enough to me :)

closed 2022-05-28 by keirf

hadess, 2022-05-28:
And I managed to rip all the Microsoft Works pour Windows 95 floppies, all of which are in DMF format except the first one:
https://archive.org/details/microsoft-works-windows-95-french

hadess, 2022-05-28:
And for my own record, the parameters needed were in https://github.com/keirf/flashfloppy/blob/master/src/image/img.c#L106 (the line marked `1.68M`). I was looking for `1680` and `DMF` so didn't find it...