keirf/greaseweazle #198: Support question: Supported drives and floppy formats? 8" specifically

VorpalBlade, 2022-05-26:
Hi,

I looked around this repo as well as the wiki and couldn't find an answer to this, nor any place to ask except the bug tracker. Sorry for that.

I'm in need of a way to read old mystery 8" floppies found in a room in the basement of the local university I work at. I do not know the format of these (there is HP and DEC equipment in the collection, what goes with what is however largely unknown, and some of it was apparently donated to it later as a retro collection, so some things may not be related at all). There are several 8" drives as well that may or may not be PC style. Thus before I buy a greaseweazle:
* Can it be connected to 8" drives? If so which ones (not all the drives I have access to have the same connector). Do I need some adapter?
* Do I need to know the format first or can I determine that after imaging to a raw flux image file?

There are also some more modern floppies to recover in the same collection, including some which I know for a fact do not work in standard PC drives (early 800k Apple ones for example). Can the greaseweazle handle those too?

Thanks in advance.

PS. It would be nice with a page listing everything supported (or possibly, if it is shorter, a list of everything unsupported).

keirf, 2022-05-28:
Yes it can be connected to an eight-inch drive but an adapter will be required as the standard eight-inch Shugart interface has 50 pins. You will also need a suitable power supply for your eight-inch drive.

You can dump to raw flux (eg SCP) and then decode using HxC tools, or my own disk-utilities project. Specific eight-inch formats can also be added to greaseweazle itself.

VorpalBlade, 2022-05-28:
Thanks. Now I just need to figure out if those drives use the standard connector next time I get back there. I know at least some of them don't.

VorpalBlade, 2022-05-29:
Thanks for the help, but consider improving the documentation, specifically with regards to what formats are supported (tested) as well as may work and are known not to work.

closed 2022-05-29 by VorpalBlade