keirf/greaseweazle #201: request: AT32F403ARGT7 support of

bestmmk, 2022-06-10:
i have a question for Greaseweazle V4 - QFP64

AT32f403RGT7 was legacy and no longer sale in the market. There is only AT32F403ARGT7 in the market

however, after programming, the board fails to work, and the LED blink for 4 times, which proves that the firmware cannot recognize the chip.

it may need firmware be improved and make recognize of the new chip

keirf, 2022-06-10:
Yes the 403A is basically compatible. However the firmware only recognises whitelisted device types, and 403A is not currently whitelisted. I'm sure this would be easy to fix. However, I currently have around 1000 assembled V4 units for sale, and the factory likely has 403 chips for another 500-1000 V4 units. Hence I am not particularly motivated to investigate 403A right now.

keirf, 2022-07-13:
This will be supported in next release.

closed 2022-07-13 by keirf