keirf/greaseweazle #204: Request: Add support for NEC PC-9800 Series floppy format

OldMadMan, 2022-07-04:
Dear developer,

Sorry, I'd like to ask if GreaseWeazle can add read/write support for NEC PC-9800 series floppy disk format.

The common PC98 HD format is: 2 sides, 77 tracks per side, 8 sectors per track, 1024 Bytes per sector, and the total capacity of the disk is 1,261,568 Bytes (often referred to as 1232KB). And, to read such 
disk, FDD motar must spin at 360rpm. 

For a 5.25-inch floppy disk or a 3.5-inch floppy disk, the above structural divisions are the same.

Thanks!

keirf, 2022-07-04:
Do you have a setup where you can test the support if I add it? It's easy to add, but I only add features that someone can test for me.

OldMadMan, 2022-07-04:
> Do you have a setup where you can test the support if I add it? It's easy to add, but I only add features that someone can test for me.

Thank you for replying me。

I have purchased GW hardware from your ebay store and assembled it with the floppy drives, and I have several such PC-9800 floppy disks to test with (well, I found software called FluxEngine can dump PC98 disks using GreaseWeazle，although I succeeded, I don't know what the rationale is.)

keirf, 2022-07-04:
Yes, FluxEngine is good! Okay, I will work on some patches and post a link here for you to test. Thanks!

keirf, 2022-07-05:
Please can you try the following tools:
https://github.com/keirf/greaseweazle/suites/7217264817/artifacts/289092911
The formats of interest to you are `pc98.dd` and `pc98.hd`. The latter is your 1232kB format.

OldMadMan, 2022-07-05:
> Please can you try the following tools: https://github.com/keirf/greaseweazle/suites/7217264817/artifacts/289092911 The formats of interest to you are `pc98.dd` and `pc98.hd`. The latter is your 1232kB format. PS: You need to be logged into GitHub for the above link to work.

Thank you so much for your hard work!

My test enviroment:

GreaseWeazle V4.
A 5.25 inch FDD (drive A) and a 3.5 inch FDD (drive B), both connected to GreaseWeazle via a ribbon cable.
My 3.5 inch FDD's motor can only spin at 300rpm.

I used the following command to read several floppy disks in PC98 HD format:

`gw read --drive B --format pc98.hd test.img`

The result of operation is not so well. GW first reads the first 154 tracks (Track 0.0 to Track 76.1) of the floppy disk normally. At this time, all valid tracks of the floppy disk should be read, but GW will still try to read Track 77.0 to Track 80.1, however these tracks are not valid for PC98 format floppy disks. The resulting floppy image was also not the correct size, implying that extra data was mixed in.

P.S. Should I use `--adjust-speed 360rpm` and `--fake-index 360rpm` to read? I've tried so, but get the same error.

keirf, 2022-07-05:
Use `--tracks=c=0-76`

keirf, 2022-07-05:
Can you tell me are all PC98 HD disks 77 cylinders only? What about PC98 DD disks?

commit 4df6992f (2022-11-15): gw: Support pc98.dd and pc98.hd formats.
Fixes #204
closed 2022-07-13 by keirf