keirf/greaseweazle #206: issue : bad sectors with the "convert" action

Telev78, 2022-07-09:
Hello,

I read some 5.25 1.5Mb floppy with this command line `gw read --drive B disk1\track00.0.raw`
I have an issue with one floppy.
When i open the file with HxCFloppyEmulator, the backup is correctly read and the track analyser don't show any bad sector.

[image]

The export function with HxC work correctly.

But, when i use this command line : `gw convert --format ibm.1200 disk4\track00.0.raw disk4.ima`, i get many bad sector...

[image]

When i check the difference between the 2 exports, i have 2 files with different hash

[image]

Why gw is not able to convert correctly this floppy ? I made a mistake ?

Thanks for your help.

keirf, 2022-07-09:
Can you zip the raw dump and upload it somewhere for me to download and look at? There's a few possibilities what it could be.

Telev78, 2022-07-09:
Here : [my dump](https://mega.nz/file/eGxgVQLS#e_TlSOPyvvhSC8825EcYoDTAtCfFNu6J1hYw1JicAjE)

keirf, 2022-07-09:
Thank you. Agreed it decodes okay in HxC tools, and also in my separate disk-utilities project. I will investigate further why greaseweazle's decoder is having problems here.

Telev78, 2022-07-09:
ok thanks for your help

commit ea8ca688 (2022-07-13): ibm: Fix bugs in raw sector decoder
Fixes #206
closed 2022-07-13 by keirf