keirf/greaseweazle #208: Firmware update fails ()

c-hri-s, 2022-07-20:
Hi,

I've downloaded Host Tools 1.1, and it tells me there is firmware update which I should apply.
This fails with the following error:

`D:\Downloads\Greaseweazle>gw info
Host Tools: v0.39
Device:
  Port:     COM3
  Model:    Greaseweazle F7 Plus (Ant Goffart, v2)
  Firmware: v0.25
  Serial:   GW002F001D3438510B36383538
  USB Rate: Full Speed (12 Mbit/s)

*** New firmware v1.2 is available
To perform an Update:
 - Run "gw update" to download and install latest firmware

D:\Downloads\Greaseweazle>gw update
Downloading latest firmware: greaseweazle-firmware-v1.2.upd
** FATAL ERROR:
File is not a zip file
`

c-hri-s, 2022-07-20:
Reinstalled into a new folder and it appears to work

closed 2022-07-20 by c-hri-s

keirf, 2022-07-20:
Yes, apologies, I changed the naming of firmware zip files slightly. This is handled by Greaseweazle Tools >= 0.40.

So, just update to latest tools (1.1) and you will find the problem is gone.

c-hri-s, 2022-07-20:
Yeah, completely my fault - I'd downloaded the new host tools, but extracted them into a separate folder. Sorry :-)