keirf/greaseweazle #210: issue : bad sectors with the "convert" action

Telev78, 2022-07-30:
Hello,

Same bug as #206  but with the host tool 1.1
convert report 1 bad sector :
[image]

HxC export fine :
[image]

data used : [link](https://mega.nz/file/KHQhHTJa#XEVmiPJqldc9mIh6wyOVfaxajIhWpQpW6nSWyM_rZD0)

keirf, 2022-07-30:
The cause is different. This time there is some spot of mould or similar around cyl 24 side 1 which is making the bitcells noisy.

Still it should be possible for Greaseweazle to do better. The PLL is being too skittish and getting thrown off by the noise.

Telev78, 2022-07-30:
it's a dump made some time ago.
The read function is also impacted by this ?

keirf, 2022-07-30:
Read to raw flux is not affected (SCP and Kryoflux stream formats). The PLL is used when converting flux to bitcells.

keirf, 2022-07-30:
The skittish aggressive PLL parameters stem from an aggressive variable-rate Amiga protection track (Speedlock). Nothing else seems to need such aggressive settings, and Greaseweazle isn't handling copy protections anyway. I've just toned the PLL parameters down by a factor of about three!

Please can you try the test build linked below (you need to be logged into github for the link to work):
https://github.com/keirf/greaseweazle/suites/7592909339/artifacts/314530353

For reference, you would now get the old PLL behaviour by specifying `--pll-period-adj=5 --pll-phase-adj=60` on the convert, read, or write command lines.

Telev78, 2022-07-30:
Thanks for your quick actions !
The test version work fine with the 5 floppies tested. The convert are done correctly.

Question : if i use gw with the command line : `gw read --raw --format ibm.1440 track.00.0.raw`
The format parameter is used only for verifiyng the dump. Withe this bug, it can report an error while the reading is correct ?

keirf, 2022-07-30:
Yes that's the idea. Your raw flux will be saved but it will be checked against the specified format and you can consider any missing sectors as a warning that your dump may not be good. But for example you would be able to gw convert your saved flux with different pll parameters.

Telev78, 2022-07-30:
ok thanks for the informations

keirf, 2022-07-30:
On further thought I realised that actually the aggressive PLL is quite good at quickly locking on to unexpected and varying bit times. The conservative PLL sure is a good fallback for noisy regular tracks though. So I've implemented both, and both will be tried (if necessary) during reads and conversions.

https://github.com/keirf/greaseweazle/suites/7593906375/artifacts/314598708

Telev78, 2022-07-31:
Testing this build looks good

keirf, 2022-08-02:
Fixed in latest release.

closed 2022-08-02 by keirf