keirf/greaseweazle #217: Failing to Install on MacOS

matthewmcneill, 2022-08-20:
I'm trying to install on MacOS.  Following the instructions in the INSTALL file.

- installed brew
- installed python3
- installed Xcode command line tools

GW installer failed: ModuleNotFoundError: No module named 'setuptools_scm'

I have checked setuptools in sinstalled:

BASH> sudo python3 -m pip install --upgrade setuptools
Collecting setuptools
  Downloading setuptools-65.1.0-py3-none-any.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 11.1 MB/s eta 0:00:00
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 63.4.3
    Uninstalling setuptools-63.4.3:
      Successfully uninstalled setuptools-63.4.3
Successfully installed setuptools-65.1.0

---[INSTALLER ERROR]-------

BASH> python3 -m pip install . 
Processing /Users/user/Downloads/greaseweazle-1.2
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      Error in sitecustomize; set PYTHONVERBOSE for traceback:
      AssertionError:
      Traceback (most recent call last):
        File "/usr/local/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/usr/local/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/local/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/setuptools/build_meta.py", line 338, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/setuptools/build_meta.py", line 320, in _get_build_requires
          self.run_setup()
        File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/setuptools/build_meta.py", line 482, in run_setup
          super(_BuildMetaLegacyBackend,
        File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/setuptools/build_meta.py", line 335, in run_setup
          exec(code, locals())
        File "<string>", line 2, in <module>
      ModuleNotFoundError: No module named 'setuptools_scm'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
BASH> greaseweazle-1.2 %

keirf, 2022-08-20:
Possibly my dependencies are a bit wrong in pyproject.toml. But anyway it seems you are missing setuptools_scm so try pip installing that? It's a separate package from setuptools itself.

matthewmcneill, 2022-08-20:
Amazing!!  That did it.  Many thanks!  

sudo python3 -m pip install --upgrade setuptools_scm

Is that worth adding to the instructions (I was installing on a new install of catalina)

keirf, 2022-08-20:
Possibly. It's pretty weird it wasn't picked up automatically.

closed 2022-08-20 by keirf