keirf/greaseweazle #218: macOS - Could not find SPS/CAPS IPF decode library

nicktones, 2022-09-03:
After installing GW on latest macOS and following the instructions to install the CAPSImage library, I get the error below when trying to write an IPF image. Can confirm that the library is installed in /Library/Frameworks.

** FATAL ERROR:
Could not find SPS/CAPS IPF decode library
For installation instructions please read the wiki:
<https://github.com/keirf/greaseweazle/wiki/IPF-Images>

keirf, 2022-09-03:
Can you tell me what the library is called under /Library/Frameworks? Is there a CAPSImage.framework/CAPSImage? Or possibly in more recent releases one or other name component is CAPSImg instead of CAPSImage. If so, it will need a fix to Greaseweazle.

nicktones, 2022-09-03:
The path is /Library/Frameworks/CAPSImage.framework/CAPSImage this is an alias to /Library/Frameworks/CAPSImage.framework/Versions/A/CAPSImage

keirf, 2022-09-03:
And what sort of a Mac is it you are running on?

nicktones, 2022-09-03:
M2 MacBook Air. Is CAPSImage Intel only?

keirf, 2022-09-03:
Fortunately not, as source code was released and FS-UAE releases binaries for macOS ARM64. You can find the download link here: https://fs-uae.net/download#macosx

I think you need to download an unzip it. And relocate the `CAPSImg.framework` subfolder to `/Library/Frameworks/CAPSImg.framework`.

You also need Greaseweazle to search for the new library name. I have put a fix on branch `macos-caps`. You can install this branch into your Python environment as follows:
```
python3 -m pip install git+https://github.com/keirf/greaseweazle@macos-caps
```

nicktones, 2022-09-03:
Perfect. I can confirm those steps work.

Thanks!!!

closed 2022-09-03 by keirf

nicktones, 2023-04-06:
Is the latest version of the tools searching for the framework? I've updated and am getting the not found error again.

commit 2f97b81d (2022-09-03): ipf, macos: Look for library under name CAPSImg

Fixes #218