keirf/greaseweazle #225: Strange red lines appear when analyzing flux files dump by GW using HxC Analyzer

OldMadMan, 2022-10-23:
Hello Keirf,

I usually use the HxC Floppy Emulator's supporting software to load the flux files dumped by GW, generate Visual Floppy Disk patterns and view them to ensure the accuracy of the capture.

Today I suddenly found that the software generated pattern has a whole red line at the edge of each track and each sector. As far as I can tell, the red line means the data is wrong, but the software still reports "0 bad sector".

These screenshot shows condition:
[image]

[image]
[image]

And strangely, these red lines only appear when grabbing a 5.25" floppy, not a 3.5" floppy. I tried to rule out various possible factors, such as replacing multiple different 5.25 floppy drives, replacing multiple data cables, grabbing different disks, and connecting the GW to a different USB port on the motherboard, but the red wire still appears .

This has not been the case since the beginning, it just happened this morning. Before that, there was no (or barely visible to the naked eye) red line in the grab result.

I would like to ask you, what is the reason for these red lines? Does their presence mean bad crawling results? What is the impact on data integrity?

thank you very much.

keirf, 2022-10-24:
These red lines mean the sectors have been written to after the track was formatted (or mastered at a factory).

The way these IBM tracks work is that individual sectors can be written at a time. Of course this creates splice points at the start and end of these writes.

So probably you happen to have looked at 5.25 disks written by a user, and 3.5 disks professionally mastered.

OldMadMan, 2022-10-27:
In fact, the floppy disk in the picture belongs to a box of completely unopened, brand-new operating system software, so I personally guess that it should not have been altered by a third party. This is a little strange.

keirf, 2022-10-27:
That is certainly mysterious. I would expect a mastered track to be written in one pass. Perhaps this got done by sector? In any case the explanation is write splices, and they are harmless.

closed 2023-01-02 by keirf