keirf/greaseweazle #229: Solved fluxmap support for IBM PC (upd765a) formats

tdaede, 2022-10-28:
A2R has a concept of a "solved fluxmap" - e.g. a flux map that has been machine generated to be perfect, not a raw read from a disk.

These can be created with gw indirectly right now, by reading to an image format and then writing back to a flux image. However, doing so loses some data that can be seen by the upd765a's Read Track command.

There are several ways to fix this. One method that I'd like to try is doing a full decode of the track (and multiple retries, if needed, to get the best possible read) but then extracting the post-PLL-recovery bitstream and encoding that to the solved fluxmap, possibly with a small amount of rotation around the index.

keirf, 2022-10-28:
To preserve gap protections etc?

tdaede, 2022-10-28:
Yes. This isn't the only way to do it, of course. This method has some downsides, too - it can't combine good sectors from separate reads, it preserves potentially useless data between sectors, etc.

tdaede, 2023-05-16:
Yeah the raw.* formats should be enough, I'll try them out and see what happens.

I was thinking that it would also be good to run the decode in this case so that the "best" of multiple tracks can be used, but I'll have to play around and figure out how much of an actual problem that is first.

tdaede, 2023-07-12:
FWIW I've been testing the raw.* formats with HFE and it's worked great so far. My current workflow then involves checking that the result is reasonable with HxCTools, but of course having a generic IBM track resolver could avoid that step (and potentially allow rereads during dumping). That would be enough to close this.

I've run into a couple of other minor issues but I'll file them separately.

keirf, 2023-07-12:
You should try the new format `ibm.scan`